mholgatem/GPIOnext #2: Alwas on

TjOsT1975, 2017-10-11:
I get these message on config:
Please release all buttons to continue
and in retropie one tab on left and it stays on.

mholgatem, 2017-10-11:
you'll have to give me a bit. I'm looking into the problem. I'll let you know as soon as I have a fix.

mholgatem, 2017-10-11:
ok, do this. 

```
cd ~/GPIOnext
wget https://raw.githubusercontent.com/mholgatem/GPIOnext/master/update.sh
git add update.sh
sudo bash update.sh
```

That should fix the problem

TjOsT1975, 2017-10-12:
Seems to work. Thank you.

mholgatem, 2017-10-12:
yep. let me know if you run into any other problems!

closed 2017-10-12 by mholgatem

TjOsT1975, 2017-10-12:
pi@retropie:~/GPIOnext $ sudo gpionext config
/usr/bin/gpionext: 6: /usr/bin/gpionext: shopt: not found
/usr/bin/gpionext: 146: /usr/bin/gpionext: [[: not found
Unrecognized Parameter: config
You can control the GPIOnext daemon using the following commands: 

- gpionext start				-> Enable + Start daemon						
- gpionext stop					-> Stop daemon until reboot or gpionext start is called
- gpionext reload				-> Reload daemon
- gpionext config				-> Run gpionext config						
- gpionext disable				-> Stop daemon/Disable auto-start on boot (type gpionext start to re-enable auto-start)						
- gpionext journal				-> Display any system output 
- gpionext test [1-4]				-> run jstest for gpionext virtual controller
-------------------------------------------------------------------------------------------------------------------------------
- gpionext set combo_delay [milliseconds]	-> Delay in milliseconds to wait for additional buttons to be pressed for combos						
- gpionext set debounce [milliseconds]		-> Delay in milliseconds to wait to avoid multiple accidental presses		
- gpionext set pins [default | pin numbers]	-> default = all available pins or comma separated pin numbers (ex. 3,5,7,11,40)					
- gpionext set pulldown [true | false]		-> set pins to use pulldown resistors instead of pull up (excludes pins 3 + 5)		
- gpionext set dev [true | false]		-> output information to console while program in use (use 'gpionext journal' to view)
- gpionext set debug [true | false]		-> output information to gpionext/logFile.txt

						
/usr/bin/gpionext: 171: /usr/bin/gpionext: shopt: not found

mholgatem, 2017-10-13:
don't use `sudo`

it's just `gpionext config`