mholgatem/GPIOnext #3: Diag

canheads, 2017-11-05:
Hello,

Thx a lot for this. It seems to work great.
Anyway when I'm playing it goes to pause mode each time I'm using to keys at the meantime (up+left for example)
Any idea how to fix this ?
Thx a lot.

canheads, 2017-11-05:
Actually it's only for player 1 up + right for some reason I can't figure out

mholgatem, 2017-11-06:
try re-running the gpionext configuration menu by typing

`gpionext config`

after that, I would also try re-running the controller config inside of retropie.

Also, you can test your virtual joystick by running the command. 

`gpionext test 1`

(1 being the player number that you want to test)

If you are still having problems, double check your wiring configuration for your buttons and get back to me.

closed 2017-11-11 by mholgatem