mholgatem/GPIOnext #5: No module named evdev

iceace86, 2017-11-21:
After the installation, when I try to config, I obtain this error:

`Traceback (most recent call last):
  File "/home/pi/GPIOnext/config_manager.py", line 14, in <module>
    from evdev import ecodes as e
ImportError: No module named 'evdev'`

I'm trying to install it on retropie (last version) on a rpi3

mholgatem, 2017-11-21:
sounds like evdev didn't install correctly. in the command line, try this:

`sudo pip3 install evdev`

if it installs, then great! restart gpionext with the command

`gpionext start`

if you get the message

> Requirement already satisfied: evdev in /usr/local/lib/python3.5/dist-packages

then we'll have to dig a bit further

iceace86, 2017-11-22:
You were right, the problem was a bad evdev installation.
Sorry for the false issue reported

mholgatem, 2017-11-24:
nah, It's no problem. Eventually I'd like to rewrite the installer so that it runs checks for this kind of stuff.

closed 2017-11-24 by mholgatem

philmck21, 2019-05-16:
I also seem to be getting this error, unfortunately for me sudo pip3 install evdev gives me "sudo: pip3: command not found"

I've tried every method I can find online to install pip3 but nothing I do seems to work.  It's as if it can't find anything pip or pip3 related in the repository or something.

I'm running Raspbian for a retropi project i'm building.

Any help you could give would be much appreciated.

mholgatem, 2019-05-17:
try this and let me know any errors that it outputs.

`sudo apt-get install python3 python3-pip python3-dev gcc`

philmck21, 2019-05-17:
Thanks very much for helping me out.  Unfortunately this is what I get ;

Reading package lists... Done
Building dependency tree
Reading state information... Done
Package python3-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Unable to locate package python3-pip
E: Package 'python3-dev' has no installation candidate

Thanks Again

mholgatem, 2019-05-18:
ok, make sure that your APT repositories are up to date:

```
sudo apt-get update
sudo apt-get dist-upgrade
```

then run
`sudo apt-get install python3-dev python3-pip`

philmck21, 2019-05-18:
Well, I got further than before.

sudo apt-get update - was already up to date
sudo apt-get dist-upgrade - this did a fair amount (I had been trying sudo apt-get upgrade no "dist-"

After the upgrade I then was able to install python3-pip, but not python3-dev.

When I try to run gpionext configure I now get a different error;

Traceback (most recent call last):
  File "/home/pi/GPIOnext/config_manager.py", line 10, in <module>
    import sqlite3
  File "/usr/local/lib/python3.7/sqlite3/__init__.py", line 23, in <module>
    from sqlite3.dbapi2 import *
  File "/usr/local/lib/python3.7/sqlite3/dbapi2.py", line 27, in <module>
    from _sqlite3 import *
ModuleNotFoundError: No module named '_sqlite3'

I tried sudo apt-get install libsqlite3-dev, but got this;
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libsqlite3-dev

If it helps I check the repo links and uncommented the bottom ones (online forum advice for someone else with a similar problem:

deb http://raspbian.raspberrypi.org/raspbian/ stretch main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
deb-src http://raspbian.raspberrypi.org/raspbian/ stretch main contrib non-free rpi

Thanks again, I know I'm a pain.

philmck21, 2019-05-18:
Also tried this;

sudo pip3 install sqlite3
Collecting sqlite3
  Could not find a version that satisfies the requirement sqlite3 (from versions: )
No matching distribution found for sqlite3

sudo pip3 install libsqlite3
Collecting libsqlite3
  Could not find a version that satisfies the requirement libsqlite3 (from versions: )
No matching distribution found for libsqlite3

mholgatem, 2019-05-19:
you have packages missing from your install, you need to install the **python3-dev** 
but it sounds like your sources list is borked, which is out of scope for this trouble-shooting.

If you are interested, your sources list is located at **/etc/apt/sources.list**

philmck21, 2019-05-20:
I checked those before I posted and both /etc/apt/sources.list & sources.list.d seem to be what they should be.

Thanks for your help, I guess I'll try wiping Raspbian & starting from scratch.

mholgatem, 2019-05-20:
ok, yeah, it seems like something must have messed up on the initial install. I think that  you are correct, a fresh install is probably the best course of action.

philmck21, 2019-05-21:
I started from scratch, no idea what happened with the first install I did but GPIOnext works perfectly now.

Thanks very much for your help & time.  Thanks for GPIOnext too, it's great and will save me a lot of extra work!!

mholgatem, 2019-05-21:
Great! Glad it's working now!