mholgatem/GPIOnext #6: GPIOnext and PWM audio

julenvitoria, 2018-02-01:
Hi, I am building a GBZ and after I install and configure GPIOnext the sound through PWM (PWM0 on GPIO #18 (ALT5) and PWM1 on GPIO #13 (ALT0)) don't work... before this it was working well.
any suggestion?

Thanks!!

mholgatem, 2018-02-05:
try running GPIOnext with the 'set pins' flag

copy and paste the following two lines into your command line:
```
gpionext set pins 3,5,7,11,13,15,19,21,23,29,31,35,37,8,10,16,18,22,24,26,32,36,38,40
gpionext reload
```
I left out pins 12 + 33 (GPIO_18 & GPIO_13 respectively).
You may need to restart your system after running the commands to get audio working.

mholgatem, 2018-02-09:
I'm going to close this issue. If you are still having problems, you can comment on here and I'll re-open the issue.

julenvitoria, 2018-02-09:
First of all thank you for your answer. Right now I can not verify the solution you gave me. You can close this issue and when I verify that the sound through pwm is functional with GPIOnext I will let you know. Thank you very much!

julenvitoria, 2018-02-15:
Hi! I just tried your suggestion but there is still some problem ... when I enter the configuration menu the audio stops working and when I try to configure the controls the pin 33 (GPIO 13) detects it at the same time as the button that I pulse.
[image]

mholgatem, 2018-02-19:
sorry, I thought that you already had the config set up. That command only affects the main GPIOnext, not the config. run the config like this:

```
sudo python3 /home/pi/GPIOnext/config_manager.py --pins 3,5,7,11,13,15,19,21,23,29,31,35,37,8,10,16,18,22,24,26,32,36,38,40
```

that should run the config and ignore the pins that you are using for pwm

julenvitoria, 2018-02-22:
Hi, I did what you told me via putty but I get an error (unrecognized arguments). Could you tell me if I did it correctly?
[image]

mholgatem, 2018-02-28:
I'm not sure what to tell you. I've tested every way that I can think of and can't replicate your problem. Have you tried just copying and pasting the command that I posted into putty?

julenvitoria, 2018-02-28:
Hi, yes, I copied an pasted the command vía putty... next weekend I will try with another pi zero that I purchased. Should I see a confirmation message after I paste the command?

mholgatem, 2018-02-28:
that is really strange. it won't be a confirmation message, but it will run the config

julenvitoria, 2018-03-04:
Hello! I have good news! I tried to make the pin configuration again with the new raspberry and this time worked perfectly. Maybe the other raspberry has some kind of problem that I do not know, it has never shown symptoms of failure in any way. Once again thank you very much for everything!

mholgatem, 2018-03-04:
cool. I'm glad that it's working because I was at a loss! Thanks for letting me know.

closed 2018-03-04 by mholgatem