mholgatem/GPIOnext #7: Lag on keypress

flagoss, 2018-03-06:
Hi !! I got GPIONext installed on Retropie. I configured 1 controller. Everything if working perfectly....exept that I am getting a lot of lag on inputs. I try setting up a Joypad and Keyboard...same issue. I am testing in Track and Field where you have to rapidely press the button to run faster....Whit GPIONext...I have a hardtime running....But with my USB Keyboard , USB Gamepad and USB Control Stick....everything is fine.

mholgatem, 2018-03-06:
USB will pretty much always be faster. That being said, there are 2 flags that you can set to speed things up. If you don't use the built in combo buttons function, Type this into the command line:

**`gpionext set combo_delay 0`**

Combo Delay is the period of time in milliseconds that the system will wait before sending controller events. This is so that you can map "A" → Button1, "B" → Button2, "C" → Button1+Button2. Without the delay its hard to trigger the "C" event because both "A" and "B" would have to be hit in the same millisecond. This will set the combo delay to 0. **By default, the combo delay is 50 milliseconds.** 

If it's still not fast enough and you have good momentary switches in your buttons:

**`gpionext set debounce 1`**

Debounce is the amount of time in milliseconds that the system will stop watching for press events from your button. Momentary switches are mechanical by nature, so when you are starting to press the button there can be one or more false connections before the actual full button press. My switches are pretty good, so I have this set to 1. **Default is 20 milliseconds**

Hope this helps!

flagoss, 2018-03-07:
OMG ... That did the trick !!!! =) 
`gpionext set combo_delay 0`
That flag did remove all delay I had. I can still use my combo since I have a standalone SHIFT key for my combos.

mholgatem, 2018-03-07:
Great! Glad that worked for you.

closed 2018-03-07 by mholgatem

mholgatem, 2019-01-03:
yeah, I had toyed with the idea of adding shift registers to the config menu but it was difficult to convey to the user why there were suddenly extra buttons. so I figured they could just leave one or two buttons unused and assign them whenever they need a shift.

fedekrum, 2019-01-06:
DON'T GIVE UP !!  ;o)

I understand what you mean, It would be confusing at least for the ones that decide to set a "shift" button. But you can just tell them at the time of the configuration that N new virtual buttons are going to be added to their device.
As we are talking about speeding up button reaction here, I wonder how fast is the python implementation?

mholgatem, 2019-01-07:
If you set everything to it's fastest (debounce=1/combo_delay 0) and have good momentary switches then there won't really be a discernible difference between that and usb on a linux machine, especially a raspberry pi (the raspberry pi doesn't even contain an real-time clock).