mholgatem/GPIOnext #8: device selection not working in config menu

djanben, 2018-03-11:
I installed _gpionext_ on retropie 4.3 on a raspberrypi 3.
I can navigate up and down in the config menu, but the item selection does not work.
Nothing happens on pressing ENTER or SPACE or the like.
I cannot exit either - had to cancel with CTRL+C.
I was logged in via _ssh_.
Unfortunately, I don't see a manual way for configuring _gpionext_.

mholgatem, 2018-03-12:
First off, what type of keyboard are you using? is it a standard keyboard or like a mini-wireless keyboard? 
Second, if you are using a standard keyboard, are you pressing the _Enter_ that is on the keypad, or the one that is just above _right shift_?

djanben, 2018-03-21:
Sorry, for the late reply. The github mails had been filtered out as spam.
I was logged in to the raspberrypi from my laptop via _ssh_.
The laptop's keyboard does not have a number keypad.
I was using the normal ENTER key above right-shift.
I had also tried to change _process_user_input()_ in _curses_menu.py_ to check for _curses.SPACE_ or _curses.ENTER_ rather than _ord('\n')_ - without success.
In the meantime I have removed gpionext and installed gpioneer and configured it via the web interface.

mholgatem, 2018-03-21:
hmm... ok. which ssh program and OS are you using on your laptop? I'll see if I can replicate the problem that you are running into.

djanben, 2018-03-21:
I'm using [Solus Linux](https://solus-project.com) with OpenSSH_7.6p1, OpenSSL 1.0.2n  7 Dec 2017.
I could probably switch back to gpionext and do some more tests on that.
However, as far as I can see, there is no debug output from the configuration menu.

djanben, 2018-03-21:
After adding some debug output I found out that the key code for ENTER is 343 which equals curses.KEY_ENTER. This seems to make sense. Please consider my corresponding pull request.

Nice work, by the way :-)

mholgatem, 2018-03-23:
ok, lets give this a shot if you don't mind. 
in curses_menu.py change line 321 from this:

`		elif user_input == ord("\n"):`

to this:

`		elif user_input = curses.KEY_ENTER or user_input == 10 or user_input == 13:`

apparently curses can receive several different values for enter depending on the terminal. I was under the impression that _ord('\n')_ handled them all, but it doesn't. Hopefully this will solve the problem. If it does, I'll make the changes in the main repository.

djanben, 2018-03-23:
I already made that change and tested it successfully. Plesse see my pull request. I have made that change in three menu files.

mholgatem, 2018-03-23:
I just saw your changes. I'll take care of it. Thanks for taking the time to test troubleshoot this with me. Let me know if there are any other problems.

closed 2018-03-23 by mholgatem