mholgatem/GPIOnext #10: GPIOnext is not recognised

tonaytan, 2018-03-29:
My Problem is, after installing a fbtp driver driver link( https://retropie.org.uk/forum/topic/295/retropie-and-waveshare-32b/36) for a Waveshare 32b screen i cannot add any buttons. If i type in evtest GPIOnext is not available. I know this seems like a problem that you shouldnt care about but adafruits parts are way to expensive and that display is working pretty fine i just cannot get evtest working . Can you please help, many of us who are Retropie are forced to use adafruits Display and its way to expensive. here is the code

Disable retrogame on startup? [y/n] (this can be undone)
-----------------
Would you like to run the configuration manager now? [y/n]y
Traceback (most recent call last):
  File "/home/pi/GPIOnext/config_manager.py", line 315, in <module>
    ConfigurationManager(args)
  File "/home/pi/GPIOnext/config_manager.py", line 85, in __init__
    gpio.setupGPIO( self.args )
  File "/home/pi/GPIOnext/config/gpio.py", line 99, in setupGPIO
    pins.append( pin(pinNumber, pull, args) )
  File "/home/pi/GPIOnext/config/gpio.py", line 59, in __init__
    bouncetime = args.debounce)
RuntimeError: Failed to add edge detection
-------------> Setup Complete!

mholgatem, 2018-03-30:
ok, so by default GPIOnext sets up all of the pins for the config and then watches for the pins to be activated. If there is a different program that is using the pins, it creates an error. But there is a flag that you can use to get around this issue. Run the config with the following command:

```
sudo python3 /home/pi/GPIOnext/config_manager.py --pins 29,31,32,33,35,36,37,38,40
```

After that, you can set the gpionext to always run with just those pins by setting:

```
gpionext set pins 29,31,32,33,35,36,37,38,40
gpionext reload
```

you should be good from there, but you may need to restart your system.

tonaytan, 2018-03-31:
thanks so much my journey has finally ended i will put a guide out and give credits to you . thanks for making an alternative to adafruits system.

closed 2018-03-31 by tonaytan

mholgatem, 2018-04-02:
no problem. Glad it helps you out!

fedekrum, 2022-12-04:
Sorry to reopening this, but I get the following:
```
pi@retropie:~ $ gpionext set pins 12,16,18,22,32,36,37,38,40
setting pins 12,16,18,22,32,36,37,38,40
Setting Flag and Restarting GPIOnext...
pi@retropie:~ $ gpionext reload
pi@retropie:~ $ gpionext config
Traceback (most recent call last):
  File "/home/pi/GPIOnext/config_manager.py", line 315, in <module>
    ConfigurationManager(args)
  File "/home/pi/GPIOnext/config_manager.py", line 85, in __init__
    gpio.setupGPIO( self.args )
  File "/home/pi/GPIOnext/config/gpio.py", line 99, in setupGPIO
    pins.append( pin(pinNumber, pull, args) )
  File "/home/pi/GPIOnext/config/gpio.py", line 59, in __init__
    bouncetime = args.debounce)
RuntimeError: Failed to add edge detection
```
So my question is what is the sent of doing `setting pins 12,16,18,22,32,36,37,38,40` if then I have to do a `sudo python3 /home/pi/repos/GPIOnext/config_manager.py --pins "12,16,18,22,32,36,37,38,40"` ?