mholgatem/GPIOnext #12: Failed to add edge detection

Manofmerr, 2018-04-30:
I'm running retropie on a raspberry pi 3 b+, and am getting the following error: RuntimeError: failed to add edge detection. i'm having a problem sshing into my pi (pretty sure it's unrelated) so i can't copy and paste the traceback. I'm currently working on getting ssh to work, if you happen to have a answer/solution without the tracebask, feel free to share. If need be, i can retype the traceback, although i'm lazy and would rather not. I'm hoping the pic will help, sorry for the quality.
[image]

Manofmerr, 2018-04-30:
And which number system do you use for the pins?

Manofmerr, 2018-04-30:
Still looking, but if it helps any, this appears to be related to pin 4, don't know if that's the issue here since i tried running it with and with out a joystick attached, and i'm pretty sure i didn't use pin 4, i know i didn't tell it to watch pin.

mholgatem, 2018-04-30:
GPIOnext uses board numbering for the pins. (ie - All the odd numbers are on the left, even on the right). 

Are you using any of the pins for anything else? Like a screen or audio? If so, you will need to run the config with the pins flag for the pins that you want to use. 
```
sudo python3 /home/pi/GPIOnext/config_manager.py --pins 3,5,7,11,13,15,etc...
```
after config, you can set the auto-run feature to always use those pins by typing
```
gpionext set pins 3,5,7,11,13,etc...
gpionext reload
```

Manofmerr, 2018-04-30:
I only had the joystick hooked up the first time, and than (because i have 1/10 of a clue) i tried to get it to work with nothing attatched, and still got that error. i have rebooted my pi a few times, but i haven't reloaded gpionext, is that something i should do even if i reboot? and do i have to type out the whole command with the path just like you did when i want to use sudo?

mholgatem, 2018-04-30:
`gpionext config`
is the same as
`sudo python3 /home/pi/GPIOnext/config_manager.py`

but if you want to use flags, then you have to type the whole thing out. I just haven't had time to implement flags into the shortcut. 

GPIOnext should already handle it, but try this:
```
gpionext stop
sudo python3 /home/pi/GPIOnext/config_manager.py --pins 3,5,7,11,13,15,etc...
```
basically the problem is that something else is grabbing control of one or more of the pins before GPIOnext does it's thing. 

What system image are you using and what software have you installed?

Manofmerr, 2018-05-01:
oops. I didn't remove retrogame after all.

Manofmerr, 2018-05-01:
Removed it, Still getting the error. Anywhere else i can look?

mholgatem, 2018-05-01:
did you reboot your system?

if so, I would look into the goodtft driver. I know that there are hdmi tft screens, but most of them connect to the gpio header. Try this to test it:
```
sudo python3 /home/pi/GPIOnext/config_manager.py --pins 37
```

Manofmerr, 2018-05-01:
not sure what that did, but now i'm in what i assume is the config screen. think that fixed it.

Manofmerr, 2018-05-01:
I need sleep, so tomorrow i'll try various buttons. may take me a little longer than i planned to play with a joystick. Electronics don't seem to like me anymore.

mholgatem, 2018-05-01:
So that command is running the config with only 1 gpio pin (pin 37). That was just to test to see if goodtft was grabbing the pins, which it seems it is. Most tft screens use pins 1-26, so if you are planning to use a gpio based screen, these are the pins that you have available to you:
29,31,32,33,35,36,37,38,40
so you will run the command like this:
`sudo python3 /home/pi/GPIOnext/config_manager.py --pins 29,31,32,33,35,36,37,38,40`

The buttons that I have listed is everything that is supported by evdev. If you have something that you want to map that isn't on the list, just use one of the generic options. It doesn't really matter, since you will map all of the buttons once you are in retropie.

Manofmerr, 2018-05-03:
I currently have my display connected by a hdmi wire. There any way to modify the driver or something else to free up those pins?

mholgatem, 2018-05-03:
According to the [goodtft github repo](https://github.com/goodtft/LCD-show), you can do this to get back to regular hdmi.
```
cd ~/LCD-show/
sudo ./LCD-hdmi
```

closed 2018-05-05 by mholgatem

degroof, 2022-07-24:
Just wanted to toss this in here in case someone else runs into this issue after installing a Raspberry Pi 4 case fan. I'd set up gpionext on a RPi 4. Worked great. 

_Then_ I added a case fan (uses PWM on GPIO14 to control fan speed). On the next reboot, all controls failed. Turns out I'd started getting the dreaded "failed to add edge detection" errors.

After a bit of digging, I tried using "gpionext set pins" to watch _only_ the pins I had buttons on. Worked like a charm. After a reboot, everything was sunshine and roses again.