mholgatem/GPIOnext #14: i2c for analog input

JediDork, 2018-06-26:
Request for adding i2c analog controllers. I can see some i2c implementation on GPIOneer but sadly it seems the work didnt progress to GPIOnext.
Can you provide details on how analog is supported in GPIOnext?

mholgatem, 2018-06-26:
the raspberry pi has only digital pins. So supporting analog controllers would require external hardware and custom code which is outside of the scope of what I intended for this project. [Adafruit has an excellent tutorial to support analog input.](https://learn.adafruit.com/reading-a-analog-in-and-controlling-audio-volume-with-the-raspberry-pi/overview)

closed 2018-06-26 by mholgatem

fedekrum, 2019-03-11:
I took a look at Adafruit's tutorial.....but my question is how to merge (if possible) that analog input to the same GPIOnext virtual joystick?

mholgatem, 2019-03-11:
The only way to do that would be to rewrite entire portions of the GPIOnext code. 
First you would need to do a major rewrite of [gpio.py](https://github.com/mholgatem/GPIOnext/blob/master/config/gpio.py) to get it to read the analog data and pass it on to [device.py](https://github.com/mholgatem/GPIOnext/blob/master/config/device.py)

Then you would need to edit [device.py](https://github.com/mholgatem/GPIOnext/blob/master/config/device.py) and either edit a joystick device to handle analog or create a new device type.