mholgatem/GPIOnext #16: is GPIOnext works with analog joystick ?

lamen, 2018-08-03:
Hi !
in first thanks for GPIOnext, it's a great software,
i need to use it with an analog joystick, i have been search on internet but nothing,
GPIOnext take only one way on the joystick, not the two way : left and right for example.
How can i work with an analog joystick ? i need a analog to digital converter like the ADS1015 ?

mholgatem, 2018-08-03:
correct. GPIOnext just reads the state of the pins, it won't read any analog values. If you are going to buy a converter, I would just cut out the middle-ware and get some of these:

[Arcade USB Encoder](https://smile.amazon.com/gp/product/B00UUROWWK/ref=oh_aui_detailpage_o04_s00?ie=UTF8&psc=1)

Those can handle digital or analog controls and you won't need to mess with GPIOnext. Best of all, they are the same price of the converters that you linked to.

mholgatem, 2018-08-05:
No problemo!

closed 2018-08-05 by mholgatem

TutelerTundra76, 2023-02-26:
Analog to digital converters do not work