mholgatem/GPIOnext #17: Everything seems to work except the Dpad

ZeroSebastian, 2018-08-29:
Everything seemed to work out, I was able to configure every button. But in Emulationstation everything works except the four directions of the Dpad. When I run gpionext test 1 also every single button is recognized. Why does this issue occur?

mholgatem, 2018-08-29:
Can you give me a little bit more info about your controller and where you are experiencing the problem?

How many joysticks do you have? how many dpads? how many buttons? Did you run the retropie controller config (after configuring with gpionext)?

ZeroSebastian, 2018-08-30:
I started configuring several times with a blank RetroPie image - without any success.
I have one Dpad, A, B, X, Y, Start, Select,R1 and L1 Buttons. When setting up the controller with gpionext config everything is recognized.
But when I run the retropie controller config every button is recognized except for the dpad. The following Pins are used for the Dpad (UP-15, DOWN-11, RIGHT-13, LEFT-16).
when I run the setpins command with my used pins nothing changes at all.

ZeroSebastian, 2018-09-04:
That is really strange... Maybe I make a video of that issue. Could there be any missing ressources?

MacGyverr, 2018-09-16:
I'm using the same version and a Pi3 and it works fine with Retropie.  GPIOs 5,6,12,13,16,19,20,26.
Thank you for your wonderful program. :)

mjasono, 2018-12-30:
Thought I might reopen this ticket as I have tried everything and searched everywhere for a solution to no avail. I am not sure what Raspberry Pi the OP was using but I am using a Pi Zero W.
I am having the EXACT same issue as OP. 

Currently, using Retropi 4.4 fresh image.
pins 8,10,16,18,32,36,38,7,11,13,31,33,37
D-PAD (U,D,L,R), A,B,X,Y, LSH, RSH, Start, Select, and a Hotkey. For both Dpad and ABXY using the PiGRRL button pad REV C (https://www.adafruit.com/product/2934). Everything detects and configures correctly in GPIONEXT. The buttons, including DPAD work in retropi-setup and raspi-config.

Test shows pressing of the UP on the DPAD. 
pi@retropie:~ $ gpionext test 1
Press Ctrl + C to exit
Driver version is 2.1.0.
Joystick (GPIOnext Joypad 1) has 2 axes (X, Y)
and 9 buttons (BtnA, BtnB, BtnX, BtnY, BtnTL, BtnTR, BtnSelect, BtnStart, (null)).
Testing ... (interrupt to exit)
Axes:  0:     0  1:-32767 Buttons:  0:off  1:off  2:off  3:off  4:off  5:off  6:off  7:off  8:off

I have tried setting it up with 0 DPADS/JOYSTICKS but it will not detect at al lin the retropie controller config menu. 

What DOES work is if I use the keyboard config in GPIONEXT menu, then it will detect all keys as a keyboard ofcourse. I just thought you may want to test this yourself. 
x   [SPACEBAR]-Check/Uncheck Item [ENTER]-Continue                             x
x                                                                              x
x   [X] ↑ UP                                                                   x
x   [X] ↓ DOWN                                                                 x
x   [X] ← LEFT                                                                 x
x   [X] → RIGHT                                                                x
x   [X] ENTER                                                                  x
x   [X] SPACEBAR                                                               x
x   [ ] LEFT-ALT                                                               x
x   [ ] LEFT-CTRL                                                              x
x   [ ] LEFT-SHIFT                                                             x
x   [ ] RIGHT-ALT                                                              x
x   [ ] RIGHT-CTRL                                                             x
x   [ ] RIGHT-SHIFT                                                            x
x   [ ] TAB                                                                    x
x   [X] A                                                                      x
x   [X] B                                                                      x
x   [ ] C                                                                      x
x   [ ] D                                                                      x
x   [ ] E                                                                      x
x   [ ] F                                                                      x
x   [ ] G                                                                      x
x   [ ] H                                                                      x
x   [ ] I                                                                      x
x   [ ] J                                                                      x
x   [ ] K                                                                      x
x   [ ] L                                                                      x
x   [ ] M                                                                      x
x   [ ] N                                                                      x
x   [ ] O                                                                      x
x   [ ] P                                                                      x
x   [ ] Q                                                                      x
x   [ ] R                                                                      x
x   [ ] S                                                                      x
x   [ ] T                                                                      x
x   [ ] U                                                                      x
x   [ ] V                                                                      x
x   [ ] W                                                                      x
x   [X] X                                                                      x
x   [X] Y                                                                      x
x   [ ] Z                                                                      x
x   [X] 0                                                                      x
x   [X] 1                                                                      x
x   [X] 2   

Thank you,

Jason (IG: Whatever_Pi)

mholgatem, 2018-12-31:
it sounds to me like it isn't so much a problem with GPIOnext, but a problem with your emulation station config, it works when you set it as a keyboard because the keyboard controls are already configured by default in ES. I would set up your GPIOnext controller the way that you want it in GPIOnext config, then go into ES config and reconfigure your controller using the prompts.

closed 2018-12-31 by mholgatem