mholgatem/GPIOnext #18: PINOUTS DEFAULT AND ON/OFF COMMANDS

andrellvs, 2018-09-07:
mholgatem, hello how are you? Greetings here from BRAZIL!

First of all, I would like to thank you. His project solved an old problem of the RETROPIE, in relation to the GPIO.

I would like you to clarify two issues that I believe have been misunderstood in GITHUB:

1) - WHAT ARE THE GPIO PINS USED TO CONTROL 1 AND 2 BY DEFAULT;
2) - HOW TO INCLUDE COMMANDS ON / OFF BY GPIO.

It would be very interesting a video of you on this. But I would be grateful if you answer here too.

mholgatem, 2018-09-08:
There aren't any default pins set up. You need to run the config utility by typing:

`gpionext config`

it will ask you questions about how many D-pads + buttons you have, then it will have you press each one in turn and watch the pins to configure it.

Turning the pi on/off. I suggest connecting your power button to pin 5 (GPIO 03). This pin can be used to power **_on_** the raspberry pi without any code. To get it to turn off the pi, you need to use the config utility and on the first screen, select the option **_Commands_**, then follow the prompts. To turn off the raspberry pi, enter the command:

`sudo poweroff`

then you should be good to go

andrellvs, 2018-09-19:
THANKS MAN! YOU ARE GREAT! -

At that moment I am also dedicating myself to creating a VGA666 CONVERTER based on the GERT666 schematics. Thank you very much for your attention!

Enviado do Outlook<http://aka.ms/weboutlook>

Closed #18<https://github.com/mholgatem/GPIOnext/issues/18>.

mholgatem, 2018-09-19:
Nice! That sounds like a fun project!