mholgatem/GPIOnext #19: Still getting "Please release all buttons to continue" after update

tristangough, 2018-10-23:
I followed the instructions here https://github.com/mholgatem/GPIOnext/issues/2
```
cd ~/GPIOnext
wget https://raw.githubusercontent.com/mholgatem/GPIOnext/master/update.sh
git add update.sh
sudo bash update.sh
```
I'm still getting the error message "Please release all buttons to continue"

It usually occurs after inputting pin 40.

mholgatem, 2018-10-23:
ok, please walk me through the steps that you are taking so I can troubleshoot. Also please describe the controller you are trying to use (number of buttons, joysticks, if you are trying to assign keys/buttons/commands, do you have anything other than your controller using the gpio). Hopefully we can get this fixed for you.

tristangough, 2018-10-23:
I start up `gpionext config`, then choose Joypad 1. The controller has 1 Dpad, Start, Select, A, B, X, Y, Left Trigger 1, Right Trigger 1, and two generic buttons.

I have a Raspberry Pi touchscreen, and an audio amp powered by the GPIO.

mholgatem, 2018-10-24:
got it. What pins are the touchscreen and audio amp using?

tristangough, 2018-10-24:
The screen is using pins 4 and 6 (5V and ground), and the amp is using 1 and 9 (3.3V and ground)

tristangough, 2018-10-25:
I used to be running Retrogame, but I turned off the daemon after I installed GPIOnext.

mholgatem, 2018-10-30:
so far I haven't been able to duplicate this issue. you may try checking to make sure that a button isn't sticking or try it on a fresh image to make sure some other software isn't doing something with the GPIO.
I'll continue testing. Was there any other software that you installed? maybe for the screen or the amp?

tristangough, 2018-11-02:
I installed a fresh Retropie image (4.4) on a Raspberry Pi 3. Still got the same problem. I'll Look into the buttons and see if they are sticking.

mholgatem, 2018-11-03:
I still haven't been able to replicate this issue. I installed on a fresh retropi image (also 4.4) on RPi3. There are a couple of flags that you could try setting that may have some bearing on the issue. You could try

`gpionext config --pins 3,5,7,8,10,11,12,[etc]`

this tells gpionext to only watch those pins during configuration (you don't need to list ground pins)

you can also try changing the debounce time, although I don't think that this is your problem

`gpionext config --debounce 100`

debounce temporarily disables the pin after a state change. So in the example above, it would disable the pin for 1/10th of a second after you press the key, then again after you release. the default amount for the configuration is 20ms which is 1/50th of a second.

mholgatem, 2018-11-05:
hmm. that's very strange. I've never encountered that problem before, but it's good to know. Was this new one also a Pi 3?

mholgatem, 2018-11-05:
I only ask because the pi3 has a B & B+ model. But I get what you are saying. I'm going to go ahead and close this issue, but feel free to reach out again if you run into any more problems. Happy gaming!

closed 2018-11-05 by mholgatem