mholgatem/GPIOnext #22: udevadm info

fedekrum, 2018-12-30:
I am trying to follow the tutorial on  https://ubuntuforums.org/showthread.php?t=1595666 to order the joystick.
In it, I found that doing

`udevadm info -q all -n /dev/input/js0`

you should get these 2 values

```
ID_VENDOR=
ID_MODEL=
```

I dont get any of those with GPIOnext, so the question is.

Is there any chance to make these 2 values available and configurable in any way ?

Thanks !

mholgatem, 2018-12-31:
The vendor and product numbers get set in [config/device.py](https://github.com/mholgatem/GPIOnext/blob/master/config/device.py#L201)
Vendor = 9999
product = 8888
But better than either of those, you can use any of the following to find the keyboard/joystick that you want (much more accurate)
```
ATTRS{name}=="GPIOnext Keyboard"
ATTRS{name}=="GPIOnext Joypad1"
ATTRS{name}=="GPIOnext Joypad2"
ATTRS{name}=="GPIOnext Joypad3"
ATTRS{name}=="GPIOnext Joypad4"
```
So, the problem that you are going to have is that you can no longer rename kernel device nodes (the tutorial that you are following is pretty old). You can test this for yourself by adding this line to **/etc/udev/rules.d/73-persistent-joystick.rules**

`KERNEL=="js?", ATTRS{name}=="GPIOnext Joypad 1", NAME="input/js5"`

then run the following commands (these are used to test udev rules against specific devices, so you'll want to hang onto them as you test things out)
```
sudo udevadm control --reload-rules
udevadm test $(udevadm info -q path -n input/js0) 2>&1
```

you will see in the output 

> NAME="input/js5" ignored, kernel device nodes can not be renamed; please fix it in /etc/udev/rules.d/73-persistent-joystick.rules:1

You can however create symlinks which will put them in '/dev'. I'm not really sure if it will help you, but I think that it is probably your best shot. Something like this:

`KERNEL=="js?", ATTRS{name}=="GPIOnext Joypad 1", SYMLINK+="js5"`

closed 2018-12-31 by mholgatem

fedekrum, 2019-01-02:
Thanks.

Sorry if I am not that good at this. Suppose that I want my only GPIOnext joystick to be always /dev/input/js0.... in kind of tutorial way, which would be the steps to make this happen even after a a reboot.

I have done what I understood from your post with no results.

Thanks again for your work

mholgatem, 2019-01-02:
[unfortunately you can't.](https://github.com/RetroPie/RetroPie-Setup/wiki/Controller-configuration-(driver-level)#udev-rules-and-device-numbering-devinputjs_x) The only thing that you may be able to do is find a script that can disable each of the usb ports, then run gpionext, then re-enable each usb port.

fedekrum, 2019-01-02:
Thanks again.
For now, I will just plug the USB stuff after Retropie starts. 

Best regards

mholgatem, 2019-01-03:
yeah, sorry I don't have a better answer. One of my biggest wishes is for linux to make it easier to assign joystick numbers.