mholgatem/GPIOnext #24: GPIOnext doesn´t works with pins 27 and 28

julenvitoria, 2019-02-02:
Hi!
 I´m trying to configure GPIOnext with pins 27 and 28 but doesn´t recognize with tehe default pins and if I try setting pins it sends me an error message like this:

mholgatem, 2019-02-03:
you can't use those pins, they are reserved for particular HAT's. It will mess with the functionality of your pi. 

[image]

closed 2019-02-03 by mholgatem

julenvitoria, 2019-02-03:
Thanks so much for the explanation!! the question was only out of curiosity because in retrogame work but I never liked to use them for what you comment

Regards

mholgatem, 2019-02-04:
no worries. It is possible to force those pins into standard input output, but it's not advised.

julenvitoria, 2019-02-05:
Thank you! Yes, I saw what is the file that contains that configuration but I prefer not to use them. Thanks again!

julenvitoria, 2019-05-04:
Hello again!! Casually I need to force the use of these two pins as standard input output. I tried editing the file constants.py and adding 27 and 28 to AVAILABLE_PINS but I can not get it to work. If it's not annoying ... could you explain me how I can make it work?

Thanks so much and regards

mholgatem, 2019-05-06:
unfortunately it's not that easy. The python module that GPIOnext uses (rpi.gpio) does not support use of pins 27 and 28 and it is beyond the scope of the GPIOnext project. If I may ask, is there a specific reason why you want to use pins 27 & 28? If you use them, it will cause problems with the raspberry pi camera and  the official (+most 3rd party) raspberry pi display.

julenvitoria, 2019-05-06:
Hi! What an inconvenience... I don´t need raspberry pi camera nor display... it is for a project of GameBoy Advance SP (https://github.com/Gameboypi/SPW). I need to use this gpio ports as normal only for buttons because the pcb is designed in that way.

mholgatem, 2019-05-07:
ok, well as stated, that is out of the scope of GPIOnext, you'll have to research to implement those on your own. It looks like he already has the buttons implemented if you download [his retropi image.](https://github.com/Gameboypi/SPW/tree/master/Retropie%20image)   
[[direct link]](https://drive.google.com/open?id=148r-MnYSLy4-oJxgjgLc_l6zv8uDW9Cg)

julenvitoria, 2019-05-08:
Yes, I know he has a premade image but it is with retrogame, I wanted to implement gpionext because it gives more possibilities, your script has the option to have more than one combo.  Don’t worry, if it's not possible there's nothing to do 😂 Thank you anyway

Regards