mholgatem/GPIOnext #30: Getting the "Failed to add edge detection" even after running remove.sh and reinstalling

benpocalypse, 2019-06-10:
I have a Waveshare 32b and a Gameboy Zero Button PCB wired up to my Raspberry Pi Zero W. Both the Waveshare and the Button PCB work separately (so I know everything is functionality correct) but I have run into an issue with GPIOnext. Some of my Button PCB wires go to "no-connect" connections that the Waveshare 32b has (specificaly 3, 5, 7, 8, and 10) so if I configured GPIOnext on a fresh install of RetroPie - everything works great. However, once I use the "LDC-Show" driver from Waveshare, GPIOnext will fail to start due to the "Failed to add edge detection" error message above. I completely understand why this error is occurring (and would love to know about any workarounds,) but my issue is with GPIOnext.

I've run the "remove.sh" done a reboot, and then run the "install.sh" only to find I can't run "gpionext config" because it fails for the same reason. I've made sure to run "gpionext stop" first, and tried to do "gpionext set pins default" and many combinations of the above...but there must be some remaining configuration laying around the filesystem that is preventing me from cleaning/reinstalling GPIOnext. Any idea what would cause this?

I realize I could just reimage the sdcard and start from scratch to not use the same pins, but that seems like overkill for something that appears to be a GPIOnext issue. Any idea what this could be?

Thank you for making GPIOnext and thanks for the help!

benpocalypse, 2019-06-11:
As another update, I wiped my sdcard, reinstalled RetroPie 4.4, installed GPIOnext, and it worked as expected. As soon as I get and install the LCD-show driver from the waveshare github, that's when GPIOnext starts giving the above errors. Not sure if this detail helps, but I'd love to be able to get this all working!

mholgatem, 2019-06-14:
sorry for the delay, been away for a bit. So the reason that GPIOnext fails after activating the lcd-show driver is that by default GPIOnext tries to watch (add edge detection) to all of the pins, but the lcd driver is already holding on to some of those pins. If you want it to run on just specific pins, there are two things that need to be done. 

1. run the config with the pins flag
`gpionext config --pins 3,5,7,8,etc`

2. set GPIOnext to run with those same pins
`gpionext set pins 3,5,7,8,etc`

hopefully that gets you up and running.

benpocalypse, 2019-06-14:
Ahhhh. I had tried doing `gpionext set pins 3,5,7,8` and running `gpionext config` and was getting the same results, but didn't realize I needed to do `gpionext config --pins xxxx` as well.

Thanks for the clarification. 

Also...I greatly appreciate you developing this tool, but maybe some more verbose error output in this situation would help prevent others from getting stuck like I did?

closed 2019-06-16 by mholgatem

mholgatem, 2019-06-30:
I have a bit of time to work on this project again. I'll try to make it a bit more verbose as well as a bit more streamlined in terms of set flags passing over to the config as well. Anything else that you'd like to see included?

benpocalypse, 2019-07-02:
Nope...I didn't run into any other issues or papercuts in my use of it.

...and thanks for doing this!