mholgatem/GPIOnext #32: Emulationstation thinks some buttons are keyboard input

Othman1992, 2019-07-06:
Hello,
I'm new in raspberry pi and don't know a lot of things.
Following a tutorial, I made a handheld gaming device similar to the one that Adafruit made.

Using GPIOnext, I successfully configured and mapped each button. And I configured them fine in emulationstation. The problem occurs when I restart the raspberry pi, some buttons get mixed up and when I re-configure in emulationstation, every time I press hold the buttons A, B and X, it says KEYBOARD instead of GPIOXX. The other buttons are fine, it's just these three.
So I have to re-configure in GPIOnext and type "gpionext start" every time I turn on the raspberry pi.

What causes this and how do I solve it permanently?

Thanks!

mholgatem, 2019-07-06:
First off, can you link the tutorial that you followed?

It sounds like you have one of two scenarios happening.

1. You configured buttons as well as keyboard keys to the same gpio pins, 
    - to fix this, run `gpionext config`
    - select `clear device`
    - select `keyboard`
    - restart
2. Retrogame (or some other gpio to keyboard utility) is enabled. To disable Retrogame
    - `sudo nano \etc\rc.local`
    - add '#' in front of the retrogame command

Othman1992, 2019-07-06:
I followed this video tutorial which introduced me to GPIOnext:
https://youtu.be/R-wAcP6wVTc

1. I tried this multiple times and it's still the same issue.

2. This file is empty, I see no retrogame command.

I have no idea what causes the issue, I'll try uninstalling and re-installing GPIOnext.
What else do you suggest?

mholgatem, 2019-07-06:
did you previously install GPIOneer? GPIOneer ran as a virtual keyboard.

Othman1992, 2019-07-06:
No, I didn't.

Othman1992, 2019-07-06:
Apparently I solved the issue.
I just uninstalled and reinstalled it. It asked me if I want retrogame on startup, I selected no (which I did the first time) and it worked.

Thanks for the help! It's still weird though.

mholgatem, 2019-07-06:
hmm... very strange. I wonder what was going on with that. Glad you were able to work it out. Happy Gaming!

Othman1992, 2019-07-06:
Thanks a lot :)

closed 2019-07-06 by Othman1992