mholgatem/GPIOnext #33: CPU usage goes very high when holding a button and pressing others

retrotails, 2019-09-22:
holding a button, then pressing other buttons several times while the first button is still held, makes CPU usage according to htop go from 0.0% to 40% on a pi zero. this seems to persist regardless of configuration flags. it also creates a lot of python processes. releasing all buttons makes CPU usage go back to zero and removes the extra processes.
[image]

mholgatem, 2019-09-22:
I don't have a raspberry pi zero, but I did set my pi 2 to single core/512mb available ram, but even so, I can't seem to replicate this.

I can however give you a little bit of insight into how gpionext works. Each time a button is pressed, it is added to a queue. A worker waits a specified time (combo_delay) to see if there are any other incoming events in order to make it a combo. Then it creates a thread for the event. The thread is self-aborting, so when it gets to the end of command, it terminates the thread. 

can you give me a little more info about your build? Let's start with OS. You can get your current OS info by typing the following two commands. The first will give us the release version, the second will give us the Kernel version

> cat /etc/os-release
> uname -a

what version of python?

> python3 --version

what config settings are you using for gpionext?
what is your configuration for your controller (keys, buttons, custom commands, etc.)

Since you are running this on a pi zero, I assume that you are making a handheld. What else do you have plugged into your gpio / (other software that accesses gpio)? There could be a conflict with applications trying to grab the same pins and gpionext not being able to properly terminate the thread.

retrotails, 2019-09-22:
@mholgatem is there any way I can disable the combo check, since I don't use it? just commenting out some lines for this specific issue is fine. setting it to 0 doesn't seem to work, and the threads persist as long as I hold the first button.
I'm running the latest retropie on kernel 4.14.98 with python 3.5.3. I disabled emulationstation and a few services, it just boots retroarch directly. I've tried all the gpionext options basically, but I noticed the issues when running the command plain, only specifying pins. I don't install it, just run the script as root.
`sudo ./gpionext.py --pins 11,36,35,15,16,32,38,7,29,13,40,37`
here's the config.db
[image]

I do have fbcp-ili9341 running to handle the LCD and a python script to handle some OS-related buttons (using RPi.GPIO), but I specifically set gpionext to use only pins that are connected to buttons, and when I disable both those such that only gpionext is using gpio, the problem persists.
all the buttons do work fine, though rarely about twice an hour one of the buttons sticks until I press it again, can happen to any of them.
only other thing I can think is relevant is, for the purposes of the LCD driver I need a fixed core_freq
`force_turbo=1`
`core_freq=400`
this has the sideeffect of pinning the CPU at 1GHz.
unrelated, I realized when trying from a clean slate recently config_manager.py doesn't have a shebang line.

mholgatem, 2019-09-23:
you can try changing the code to ignore the combo press; but I don't know if that will fix anything because it is still going to run each event as a thread. If you want to give it a shot, you need to edit config/device.py starting at line 263

change it from

```
		while self.queue:
			try:
				currentEvent = self.queue[ 0 ]
				currentBitmask = currentEvent.bitmask
				for event in self.queue[ 1: ]:
					# check if button is part of combo press
					if currentEvent.bitmaskIn( event.bitmask ):
						# remove currentEvent
						self.queue.pop(0)
						break
				else:
					# run the method
					self.DEBUG( 'Press ' + currentEvent.name )
					threading.Thread( target=currentEvent.press ).start()
					if self.queue:
						self.queue.pop( 0 )
					time.sleep( self.comboDelay )
			except IndexError:
				pass
```

to something like:

```
		while self.queue:
			currentEvent = self.queue[ 0 ]
			threading.Thread( target=currentEvent.press ).start()
			if self.queue: self.queue.pop( 0 )
```

Note: this code is entirely untested!

retrotails, 2019-09-23:
@mholgatem hmm, that did indeed make no difference.
I think I'm going to look at alternatives at this point, but I can test more if you have anything else to try.

mholgatem, 2019-09-23:
yeah, that is very strange. I tried setting my core frequency and still can't replicate it. I'm not sure what is going on. 

You could use the older version that is called [GPIOneer](https://github.com/mholgatem/gpioneer). It doesn't have all the bells and whistles (it emulates a keyboard keys and the config is pretty bare bones), but I know that it works well with retropie and doesn't use threading.

mholgatem, 2019-09-23:
In any case, good luck!

closed 2019-09-23 by mholgatem