mholgatem/GPIOnext #34: Is there a way to hold a button and it fired the command multiple times?

stazna01, 2019-10-25:
I have GPIOnext working great with two buttons (thank you!), one for volume up and one for volume down. I'd really like to be able to hold the volume buttons and they fire the increase or decrease volume commands numerous times. Is this possible with GPIOnext?

mholgatem, 2019-10-25:
it's not currently possible but I like the idea. I'm going to be away for a bit, but hopefully I can get it implemented when I get back. The only current workaround that I can think of at the moment would be to use your buttons to run a script. The script could then check to see if the gpio pin is high or low and run your command accordingly until the pin state changes.

ooskay, 2020-02-02:
Hi All,
I really would like to see hold feature. I could use volume down button for mute with long hold/press. Thanks advance.

fedekrum, 2020-02-10:
> The only current workaround that I can think of at the moment would be to use your buttons to run a script. The script could then check to see if the gpio pin is high or low and run your command accordingly until the pin state changes.

This is the way to go !!!
It may be more interesting to have examples no how to do this to be applied in any case by newbies like me.

wmoecke, 2020-05-13:
I did something similar quite easily: I had a button configured to quit emulationstation and safely shut down the Pi, but I wanted some control, i.e., to keep from "accidentally" shutting down the Pi in mid-game. So I wrote a simple python script to count for how long the button was pressed (aka, only fire the action when the button was held down for xx seconds):
`#!/usr/bin/python
import RPi.GPIO as GPIO
import time
import os

GPIO.setmode(GPIO.BOARD)
GPIO.setup(3, GPIO.IN, pull_up_down = GPIO.PUD_UP)

count = 0
shutdown = False
while count < 3:
    shutdown = not GPIO.input(3)
    count += 1
    time.sleep(.5)

if shutdown:
    os.system('sudo killall emulationstation && sleep 0.5 && shutdown -h now')
`
The above script is set to run from the GPIONext command, such as `python /home/pi/button_shutdown.py`.

It was natural to me to just go ahead and write this to make my button behave as I wanted it to. Didn't even think of specifically asking for a feature like this; IMHO it's implicit when you the devs add a custom external command feature for you.

wmoecke, 2020-05-13:
On second thought, I think that the debounce setting could be extended in order to take a second (optional) argument as the pin # for the button we want to set a specific debounce time for. This would eliminate the need for my first script (count for how long a specific button was held down).

closed 2020-06-15 by mholgatem