mholgatem/GPIOnext #35: No Gamepad detected in emulationstation

R3DPanda1, 2019-10-31:
Hi, I am trying to make a small 4 player NES emulator with GPIO gamepads using RetroPie on a Raspberry Pi Zero W. I only have 4 buttons at the moment, so I made a D-pad and set up GPIOnext. It works when I type gpionext test 1, but I can't get it to work in the emulationstation. I already tried the reload, start and stop commands, making a clean install of RetroPie, configuring 4 gamepads with only an A button on each.
My question is how can I get it to work in emulationstation and how does the daemon work? Do I need to run gpionext start or gpionext reload on boot before emulationstation? 
A weird thing that might be important is that after I reboot using the RetroPie's menu my keyboard stops working too.

mholgatem, 2019-10-31:
emulation station doesn't recognize controllers unless they have a D-pad or joystick _and_ at least 2 buttons. Add an extra button to your setup and it should recognize it. If I remember correctly, it doesn't matter what the buttons are.

`gpionext start` does 2 things:

1. registers the service with the service daemon (systemctl)
2. starts the service

from there, the service will auto start every time that you boot

`gpionext reload` does 3 things:

1. temporarily stops gpionext
2. reloads the gpionext service into the service daemon (systemctl)
3. restarts gpionext

`gpionext disable' will disable the service from running at startup by un-registering it with the service daemon (systemctl)

I don't know why your keyboard stops working, I doubt that gpionext is doing it though.

R3DPanda1, 2019-10-31:
Thanks a lot! It worked.

closed 2019-10-31 by R3DPanda1