mholgatem/GPIOnext #39: Add deadzone config option?

wmoecke, 2020-05-13:
I notice that there is way too much sensitivity for the direction pad. It really gets in the way of a good user-experience (even navigating through ES menus is a pain - playing games such as Pacman becomes a nightmare sometimes, as the buttons seem to be doing whatever they want to do, due to their extreme sensitivity).

An easy fix for this is to add a specific config for setting deadzone values. By deadzone, I mean the threshold where the daemon detects a transition from "center" to any direction. The greater this value, the more the direction pad must be moved to any direction. It's something quite alike the debounce setting, but for the direction pad. 

Thank you.

mholgatem, 2020-05-13:
This would be nice, but unfortunately the raspberry pi gpio do not support analog input. So they can't see how far from center a direction is being pushed, they only see off or on. Altering the debounce threshold to a higher value can help with menu navigation as it will help filter out false positives. There is a way to emulate analog input with hardware on the raspberry pi, but that is beyond the scope of this project. Thanks for the suggestions though! Keep'em coming!

closed 2020-05-13 by mholgatem

wmoecke, 2020-05-13:
This is not analog, I'm using a DPAD. It is the DPAD that's causing trouble due to the lack of a deadzone.

mholgatem, 2020-05-13:
Got it. But when I'm talking analog, I mean an analog signal, as opposed to a digital signal. 

Let's use your dpad as an example. If it used an analog signal, the further from center that you pushed it, the higher voltage the signal would be. Thus, you could say, "ignore all voltages that are less than 0.01" to create a dead zone where the dpad wouldn't respond. 

Unfortunately, the raspberry pi is digital only and can only discern "off" or "on" (0 or 1). So as soon as that analog signal comes to the pi, it can't see how far from center it is, only that it's not at center. So there's no way to set a dead zone. 

With some simple hardware and clever programming, you can fake an analog signal by pulsing the digital pin. The smaller the delay between pulses, the further from center you would say you are. And thus, you could measure the pulse width to determine your dead zone.