mholgatem/GPIOnext #40: Missing an uninstall option

tommythorn, 2020-05-21:
Thanks for making this, but there should be an equally easy way to uninstall it.  Now I unfortunately have to understand the code to undo the installation.

I had seen it suggested over Adafruit-Retrogame, but it was not immediately clear that it couldn't be configured on its own but requires being run with Retropie.  The "gpionext config" also allows setting the target keys, but there's confusingly no way to either manually set the mapping or have it auto-detect while in configure.  I recommend that you make that clear in the README as it certainly wasn't obvious to me.

mholgatem, 2020-05-21:
I am unsure what you mean by "manually set the mapping or have it auto detect." Are you talking in regards to retropie configuration?

tommythorn, 2020-05-21:
I have switches (joystick and buttons) tied to GPIOs; I'm talking about setting the correspondence between my switches and the GPIOs.  In Adafruit-Retrogame you configure the explicitly (originally by changing the source).  GPIOnext claims it "auto-configures". It's unclear how that work or when it's done.  I couldn't get anything, not even in the debug log.  It could be that it's related to being a Raspberry Pi 4, but I didn't see anything in the documentation about how events would show up or how exactly the auto-configuration would work.

mholgatem, 2020-05-21:
Did you run the config utility after installation? 
`gpionext config`

tommythorn, 2020-05-21:
It's run automatically and I also ran it manually many times, wondering if I was missing something.  I picked the first joystick option (this is for an arcade) and checked the first 13 consecutive boxes.  However at this point, the only option appears to go back to the previous menu and then go out.  Am I missing something?  At no point did anything I didn't on my switches make any difference nor did I have an opportunity to specify the switch - gpio mapping.

mholgatem, 2020-05-21:
You probably missed it after scrolling down to make your button selections, but at the top it says to press 'enter' to continue. That should bring you to a screen that looks like this:

[image]

it will run through 4 directions for each Dpad/joystick selected, plus 1 for each button.

tommythorn, 2020-05-22:
Well, that's embarrassing and thank you for your patience with me throughout this.  It works perfectly.    (Can you really set up 4 joypads with 4 joysticks each?)

I guess I won't be needing the uninstall option now :)

closed 2020-05-22 by tommythorn

mholgatem, 2020-05-22:
it's no problem. that's what I'm here for. 

so yes, in theory you could. but I think that in practice you would run out of gpio pins!

I originally thought about doing an uninstall option, but then I lost interest. ha!