mholgatem/GPIOnext #41: GPIOnext Giving so much errors + not recongizing

KredoHere, 2020-06-08:
Hello, i am making micro console on raspberry pi zero W. I want to set up virtual keyboard. Once i am configuring it up, emulationstation can't recongize it. In system journal i am getting errors like this:
`Jun 08 18:17:54 retropie systemd[1]: gpionext.service: Main process exited, code=exited, status=1/FAILURE
Jun 08 18:17:54 retropie systemd[1]: gpionext.service: Failed with result 'exit-code'.
Jun 08 18:17:55 retropie systemd[1]: gpionext.service: Service RestartSec=1s expired, scheduling restart.
Jun 08 18:17:55 retropie systemd[1]: gpionext.service: Scheduled restart job, restart counter is at 74.
Jun 08 18:17:55 retropie systemd[1]: Stopped gpionext-daemon.
Jun 08 18:17:55 retropie systemd[1]: Started gpionext-daemon.
Jun 08 18:17:57 retropie python3[2348]: Traceback (most recent call last):
Jun 08 18:17:57 retropie python3[2348]:   File "gpionext.py", line 116, in <module>
Jun 08 18:17:57 retropie python3[2348]:     GPIOnext( args )
Jun 08 18:17:57 retropie python3[2348]:   File "gpionext.py", line 54, in __init__
Jun 08 18:17:57 retropie python3[2348]:     gpio.setupGPIO( self.args )
Jun 08 18:17:57 retropie python3[2348]:   File "/home/pi/GPIOnext/config/gpio.py", line 99, in setupGPIO
Jun 08 18:17:57 retropie python3[2348]:     pins.append( pin(pinNumber, pull, args) )
Jun 08 18:17:57 retropie python3[2348]:   File "/home/pi/GPIOnext/config/gpio.py", line 55, in __init__
Jun 08 18:17:57 retropie python3[2348]:     GPIO.setup(number, GPIO.IN, pull_up_down=pull)
Jun 08 18:17:57 retropie python3[2348]: ValueError: The channel sent is invalid on a Raspberry Pi
`

mholgatem, 2020-06-08:
it's just one error (ValueError). Basically it's saying that it is receiving an invalid value for one of the raspberry pi gpio pins. Did you run through the GPIOnext configuration first (if not, run `gpionext config`)? how many keys/buttons are you configuring?

KredoHere, 2020-06-09:
yes, i ran gpionext config
i am using 8 pins currently, but i want use 13

mholgatem, 2020-06-09:
ok, can you send me the config database located in
*~/GPIOnext/config/config.db*
you can send it to transatlantic.contact@gmail.com

I'll take a look and see if I can figure it out

KredoHere, 2020-06-09:
I realized something. Do i need to configure numbers of gpio pins or the
header pins?

mholgatem, 2020-06-09:
I think that you are running into the same problem that someone else was having last week. While running gpionext config, make your key selections with "Spacebar", when you are ready to continue, press "Enter". That will bring you to a screen that looks like this:

[image]

From here, it will run through the keys that you selected, prompting you to press and hold a button or combination of buttons. When you do that, it will read the state of the pins to see which ones that you are activating and it will map that key to those pins.

KredoHere, 2020-06-10:
Yup, i did it but it shows pin numbers ant there are header pins not gpio pins. Should i specify header pins to watch?

KredoHere, 2020-06-14:
Ok. thank you! I was specifing GPIO pins not header pins. I will check it tonight.

KredoHere, 2020-06-15:
O, tested it... And it works!!! Thank you for fast reply!

closed 2020-06-15 by KredoHere