mholgatem/GPIOnext #42: Button not detected during "Hold [button name]" in config

FreeTacoz, 2020-08-02:
Hi,

I just installed GPIOnext and everything appears to have installed correctly. I run the config manager, select Joypad 1, select 0 sticks, select Start, then press enter. The screen instructs to Hold Start Button but does not recognize the button press. I've got my button wired as in [this tutorial](https://raspberrypihq.com/use-a-push-button-with-raspberry-pi-gpio/) and set GPIOnext to pulldown. I verified the button is wired correctly by running GPIO.input(10) and checking for 0's and 1's. I have tried moving the input wire to other GPIO ports just in case, but have not been able to setup this button yet

I am using a Raspberry Pi 4 running RetroPie 4.6. Button is wired VCC -> 1k resistor -> button -> GPIO board pin 10

mholgatem, 2020-08-02:
First off, Thanks for being so thorough with your summary. It really speeds up troubleshooting. 

I think that what you are probably running into is a problem that I didn't really document. When you set the flags (**`gpionext set pulldown`**, for example). it is only setting the flags for the systemctl daemon. So you probably just need to run the config like this,
**`gpionext config --pulldown`** and it should recognize it. Alternatively, you can also set pulldown back to false and then wire your buttons as such: gpio board pin → button → any ground pin. The gpio pins are normally high and get pulled low when the button makes a connection to ground. This avoids needing a resistor. The pulldown method is just in case someone had some custom pihat that required it to be pulled low.

FreeTacoz, 2020-08-02:
Both methods worked! Thank you for the quick reply!

closed 2020-08-02 by FreeTacoz

mholgatem, 2020-08-02:
No problem. Hope this works out for you!