mholgatem/GPIOnext #43: GPIOnext crash

pultronic, 2020-08-08:
Hi all
I installed GPIOnext, configured joysticks and buttons for 2 players (26 entries in total; 1 joystick and 10 buttons for player 1 and 1 joystick and 8 buttons for player 2).
I also tweeked a little with combo_delay and debounce values to get a better gameplay but I don't remember the last numbers I set. Is there a way to check?
I quicky noticed in emulationstation menus that 1 single button press jumps 2 lines not always but happens alot. Some times it get stuck and keeps scrolling in one direction.
Here you can see the system output when I pressed 3 diferent buttons (1 click each).
[image]

But in game happens a worst cenario. In games that you need to keep pressing a button, for example a racing game that you need to keep pressing the accelarator, all buttons stop working after pressing the same button for 15 or 30 seconds.
This is the error I got:
[image]

I'm using a Raspberry Pi 3b+, running retropie 4.6. All gpio are connected this way: GPIO pin -> microswitch -> ground

mholgatem, 2020-08-09:
if you open the file /lib/systemd/system/gpionext.service. You can see all of your currently set values. There is no way to see previous values though.

default values for debounce and combo delay are:
--debounce 1 
--combo_delay 50 

from that error, it looks like your system hit the cap for creating new threads (~9000). Every time that a button is presses, a new thread is started. it runs for as long as combo_delay is active or until the button (or buttons, of a combo) are released.

pultronic, 2020-08-09:
thanks for the quick reply.
So this is my current values:
[image]

any solution to stop this gpionext error?

mholgatem, 2020-08-10:
I'm not sure what would be causing that. I would try disabling combos by setting combo_delay to 0 and see if you are still having problems.

closed 2021-08-30 by mholgatem