mholgatem/GPIOnext #44: Joystick Issue

Walli1223334444, 2020-09-30:
I wired up some buttons and a joystick. When I try to configure the joystick it doesn’t not recognise it when I hold it down or right even though it finds the pins when I hold up and left. Otherwise there haven’t been any issues.

pieter16, 2020-12-30:
Same issue here. Recalbox sees my 2 joysticks and 17 buttons, therefore it’s not a hardware issue. With retropie, GPIOnext will see everything but the down on my joystick1 - pin 11. No idea on how to solve that :(

mholgatem, 2020-12-30:
If it worked in recalbox, then the issue lies in retropie. I would guess that retropie has some module that is being activated and is grabbing that pin so that gpionext can't use it. Perhaps a touchscreen module or controller module?

pieter16, 2020-12-30:
Thanks for your answer, and for your great job. I wasn't clear enough sorry: my hardware was recognised directly by RecalBox with the embedded driver (mkjoystickarcade), not GPIOnext. Btw, for additional information I'm using a RPI4. Will try again a fresh install of Retropie with GPIOnext.

Cazna52, 2021-08-09:
Hi. I am having the same issue with GPIOnext. Which file/s could I check to see if other modules are taking pins?

closed 2021-08-30 by mholgatem