mholgatem/GPIOnext #45: Run commands

nasp2000, 2020-10-25:
Hi all

I have used gpionext it great sucess and most it simple programmable buttons like:
 sudo halt 
or volume+ or down:
/usr/bin/amixer -q -c 1 sset Headphone 4dB- unmet no cap

But now i am stuck it running scripts, for example "/home/pi/screen.sh",  programmable with or without sudo it full paths and the script simple don´t run. The script are mark it executable and i test the script first runninh ./screen.sh.

What is wrong ? How i put the script working it a button from gpionext...

mholgatem, 2020-10-26:
You need to specify the direct path to your bash interpreter. It should be something like `/bin/bash /home/pi/screen.sh`

closed 2020-10-26 by mholgatem

nasp2000, 2020-10-26:
Half easy the solution.... i tried to run like /bin/bash /home/pi/screen.sh but don´t work. Only works it user pi
like this:
su -c "/bin/bash /home/pi/screen.sh" pi

mholgatem, 2020-10-26:
I think that if you create a group and add root and the file to the group, it should work.
```
groupadd gpionext
usermod -a -G gpionext root
chown :gpionext /home/pi/screen.sh
```
(create the group gpionext
add root to the gpionext group
change the group-owner of /home/pi/screen.sh to gpionext)