mholgatem/GPIOnext #48: Raspberry pi 3 --  RPi.GPIO error

Tailslide, 2021-01-13:
Install Complete!
Would you like to run the configuration manager now? [y/n]Install Complete!
Would you like to run the configuration manager now? [y/n]y
Traceback (most recent call last):
  File "/home/pi/GPIOnext/config_manager.py", line 15, in <module>
    from config import gpio, menus, SQL
  File "/home/pi/GPIOnext/config/gpio.py", line 2, in <module>
    import RPi.GPIO as GPIO
RuntimeError: This module can only be run on a Raspberry Pi!
-------------> Setup Complete!

Tailslide, 2021-01-13:
This solved it for me:

sudo apt-get purge python{,3}-rpi.gpio
sudo -H pip2 install --upgrade pip
sudo -H pip3 install --upgrade pip
Install RPi.GPIO from updated pip:

python -m pip install RPi.GPIO
python3 -m pip install RPi.GPIO

closed 2021-01-13 by Tailslide

Devenhulk123, 2023-07-20:
Hey I have sorta the same problem and I'm wondering if you could help me this is what it says 
[image]
I copied an
[image]
D pases these codes