mholgatem/GPIOnext #54: My pi doesn't want to install evdev, saying there are errors when trying to install it.

DockPo234, 2021-09-09:


mholgatem, 2021-09-10:
That's unfortunate. Can you provide the errors that you are getting?

DockPo234, 2021-09-14:
It took me forever,  but I got it to reinstall it.  Thanks for being there
though! Not many creators respond and are ready to help.

mholgatem, 2021-09-14:
No problemo!

closed 2021-09-14 by mholgatem

mrjjenkins, 2021-10-07:
what did you do to make it work?  I am having a hell of a time with the evdev error

DockPo234, 2021-10-08:
I don't know if this will help, but I plugged in the ethernet cable to my
Pi 3B, ran 'sudo apt-get install python3-pip'. After that, I ran 'sudo pip3
install evdev'. Depending on the python version you're using, you may want
to run the previous commands without the 3. Let me know if this works, and
I can try to help you further if needed.