mholgatem/GPIOnext #55: RetroPie - config_manager seems to throw an exception

ianoxian, 2021-09-22:
Hi!
I am using the latest pre-made image of RetroPie, version 4.7.1. , for the Raspberry Pi 3.
I purchased the Joy-It GameStation a while ago and am trying to install GPIOnext as the successor of GPioneer .

However, wihle the installation is successful, I cannot get the config manager running - see the error message below.
Are there any prerequisites that might need attention so that Python3 is able to communicate with the button controller of the gamestation? Is there anything that I am missing or that I can do to further assist you in debugging this?

Best regards
Frank

```
root@retropie:/home/pi/GPIOnext# python3 config_manager.py
Traceback (most recent call last):
  File "config_manager.py", line 315, in <module>
    ConfigurationManager(args)
  File "config_manager.py", line 85, in __init__
    gpio.setupGPIO( self.args )
  File "/home/pi/GPIOnext/config/gpio.py", line 99, in setupGPIO
    pins.append( pin(pinNumber, pull, args) )
  File "/home/pi/GPIOnext/config/gpio.py", line 59, in __init__
    bouncetime = args.debounce)
RuntimeError: Failed to add edge detection
```

TheReal1604, 2021-09-28:
Hey Frank,

take a look at: https://github.com/mholgatem/GPIOnext/issues/30 Maybe this can help? 

It looks like if there are already gpio pins in use, the library got an issue with watching these pins and set the edge detection. You can use GPIOnext with specific configured pins as state in the above issue:

`gpionext config --pins 3,5,...`

mholgatem, 2021-10-04:
@ianoxian, Were you able to figure out your problems?

mholgatem, 2021-10-07:
@ianoxian - I'm going to close this issue, let me know if you are still having problems and I can re-open it.