mholgatem/GPIOnext #56: Must reload gpionext to make it work on some html5 games

Oneil974, 2021-10-03:
Hello,

So first thank for your work I’ve searched for a while to found a library that could make my Game Hat working with webgl games.

But I have a strange issue that occurs, in an html5 api gamepad tester it works perfectly but in some webgl games I must stop and start gpionext once to make it works.

I tried to reload it with a script at startup but it doesn’t works, maybe my script is wrong.
I use chromium-kiosk and when I stop and start it from ssh command as pi user it works fine. 

Have you an idea of how I could bypass this?
 Thanks

mholgatem, 2021-10-04:
This is a peculiar one. My first guess would be that your webgl games are looking for a "device connection" event or need something else to load before gpionext. Can you post the script that you used to restart it? How is your script being run? (rc.local? cron job?)

I would try editing `/lib/systemd/system/gpionext.service` to get gpionext to run later. 
You can copy and paste this over the original file or just change *simple* to *idle* in the original. This is just a simple workaround to get it to run *later*, there's a more complex way to get it to run *last* if we can determine that is needed
*(notice the type changes to idle)*
```
[Unit]
Description=gpionext-daemon

[Service]
Type=idle
WorkingDirectory=/home/pi/gpionext
ExecStart=/usr/bin/python3 -u gpionext.py --debounce 1 --combo_delay 50 
Restart=on-failure
RestartSec=1
ExecReload=/bin/kill -HUP $MAINPID

[Install]
WantedBy=multi-user.target
```

Oneil974, 2021-10-05:
Oh thanks for your answers.
I tried idle in type but it doesn’t work. Yes I’m trying to reload it with rc.local but it look like to run before chromium launch because when I had a sleep in it deferred chromium launch. I think you are right with event listener, that would be why gamepad tester work without restart. Another thing, after restart gpionext manually, even if I restart chromium or refresh the page, everything works good.
If you have another idea to start it later?
I’m creating a UI with nodejs and I think I’ll try to make it restart by nodejs when the UI would be loaded and then maybe it could works.
Does the user used to launch script take importance?

mholgatem, 2021-10-05:
Try this:
edit /lib/systemd/system/gpionext.service

Just above the line that reads: `WorkingDirectory=/home/pi/gpionext` 
add the line: `ExecStartPre=/bin/sleep 20`

That should cause it to sleep for 20 seconds before running gpionext.

Oneil974, 2021-10-05:
Doesn’t works at all :(
Is there a difference between service starting commands and « gpionext stop » then « gpionext start » ?

mholgatem, 2021-10-05:
no, `gpionext stop` is equivalent to `sudo systemctl stop gpionext`

`gpionext start` is equivalent to
```
sudo systemctl enable gpionext
sudo systemctl start gpionext
```
and
`gpionext reload` is equivalent to
```
sudo systemctl stop gpionext
sudo systemctl daemon-reload
sudo systemctl start gpionext
```
From what it sounds like, gpionext is running, just not being picked up by your games until it's reloaded. You could try just wiring and extra button (or use gpionext builtin button combos) to send a custom command that would reload gpionext.

Oneil974, 2021-10-06:
Hey nice idea it works, use combo button with command (in my case start+select) make the service reload with this command :
su - pi gpionext reload 
But for a strange reason it take about a minute (or more) to restart…
In ssh command with pi user it restart instantly

mholgatem, 2021-10-07:
cool. I'm going to close this issue, since we seem to have found a passible workaround but I'd be interested to know what the JS gamepad dev has to say.

closed 2021-10-07 by mholgatem

mholgatem, 2021-10-07:
You may be able to create a udev rule to account for this, though I don't know specifically what it would look like. 
As far as on my side of the software, evdev doesn't have any way to set that as far as I can recall. You can try [checking the docs]((https://python-evdev.readthedocs.io/en/latest/apidoc.html#api-reference)).
Do you have any other peripherals plugged in? is the screen touch sensitive? gpionext should just be listed as the first available controller.