mholgatem/GPIOnext #59: Gpionext registering double input

wheelbarrow-of-mellons, 2021-12-23:


wheelbarrow-of-mellons, 2021-12-23:
I am having an issue with gpionext where every time I press a button it registers it as two button presses.  This makes it difficult to navigate the menu or do anything.  The buttons are configured to keyboard presses, and I am not sure if it had the same problem with controler configuration.

mholgatem, 2021-12-25:
You probably need to adjust the debounce setting. 
From command line, try this:
`gpionext set debounce 50`
This will set the debounce rate to 50, meaning once a button is triggered, it won't trigger again for 50 milliseconds. You can go all the way up to 250 without noticing much lag. Any higher and you may start to notice missed button presses.

wheelbarrow-of-mellons, 2021-12-27:
I tried adjusting the denounce setting but it doesn’t seem to fix my problem.  I found that it is only registering double input on the main retropie menu, and that the it is perfectly fine in the command line and the roms.  Thanks for the help.

closed 2022-01-21 by mholgatem