mholgatem/GPIOnext #60: No module named evdev

Jackslayerr, 2022-01-02:
I’ve been trying to get GPIOnext working for hours and when I try to configure GPIOnext it says there’s no module names evdev. I’ve seen the issue posted multiple times in this project but I tried everything from all of them to try to fix this but nothing works. A couple things I’ve noticed was that while downloading something it said there was a error 404 link for the and that I was missing some things. I’ve also seen some people say a stock retropie version works, is that just an older version? Because I tried to use and older version but it didn’t work.

merwok, 2022-01-02:
> while downloading something it said there was a error 404 link

> I was missing some things

Please give useful details to make bug reports actionable!
What commands you types, what full error message you got.

Jackslayerr, 2022-01-02:
> > while downloading something it said there was a error 404 link
> 
> > I was missing some things
> 
> Please give useful details to make bug reports actionable! What commands you types, what full error message you got.

Thanks for the Reply! I got the error running the command
“sudo apt install python-pip” in an attempt to fix the issue. Here’s the output, towards the bottom is where it gives me the 404 error

[image]

merwok, 2022-01-02:
Have you tried one of the two suggestions in the last line of the error message?

Jackslayerr, 2022-01-02:
The first suggestion didn’t do anything because apparently my access was denied and I didn’t understand the second suggestion.

[image]

merwok, 2022-01-02:
You have to run system administration commands as root (using sudo)

Jackslayerr, 2022-01-02:
Ok that time it worked (Sorry I’m a bit a new to this stuff) I’m still getting that 404 message when trying to install python. What was the second suggestion?

merwok, 2022-01-02:
Can you show that error message?

Jackslayerr, 2022-01-02:
> Can you show that error message?

[image]

merwok, 2022-01-02:
Actually, there is an `install.sh` script in this repo that does apt update and more — can you use it?

Jackslayerr, 2022-01-02:
> Actually, there is an `install.sh` script in this repo that does apt update and more — can you use it?

I thought I tried it but what’s the command?

merwok, 2022-01-02:
It is an executable script, so `./install.sh`

Jackslayerr, 2022-01-02:
This is what I got 

[image]

merwok, 2022-01-02:
It’s in the gpionext repository, so you have to run that command from your clone of it

Jackslayerr, 2022-01-04:
> It’s in the gpionext repository, so you have to run that command from your clone of it
> Follow https://github.com/mholgatem/GPIOnext#how-to-install

I’ve tried everything but still nothing. I heard someone say they just got a stock version of Retropie. What other image could I get?

Jackslayerr, 2022-01-05:
So I was able to finally get it working. I went and updated Retropie within the OS and a bunch of stuff downloaded and I tried GPIOnext again and it worked! I hope I helped other people with this issue.

closed 2022-01-05 by Jackslayerr

mholgatem, 2022-01-09:
Hey, I'm glad that you were able to get it working. Sorry I was unavailable for the last week.

For anyone that comes across this problem in the future, my guess would be that the Debian/Ubuntu distribution that your retropie image used was old. Ubuntu gives it's releases a shelf life. After that, it migrates its packages to a new location to ensure that you aren't mixing old releases with new. You can typically fix this by updating the current distribution using the following command:

`sudo apt-get dist-upgrade`

or you can update your packages list with the new sources by editing:

`/etc/apt/sources.list`

Happy Gaming!

naenaejeff, 2023-12-10:
its not working for me, and that file is read only, im not sure how to edit it? any help?

merwok, 2023-12-10:
To get help, you need to say what you did (copy the exact command) and what you got (copy the full error message)
Otherwise you are asking for mind reading 😅

Billardgates, 2026-01-29:
ive done everthing updated retro pi and re download retro pi as a whole but nothing. it sill gives me this error after i clone the GPIOnext files and install. it gives me quite a few error not found messages while downloading. but then it says install complete would you like to run the configuration manager now? yes then it spits back this 
traceback (most recent call last):
  File "/home/pi/GPIOnext/config_manager.py", line 14, in <module>
     from evdev import ecodes as e
ModuleNotFoundError: No module named 'evdev'

what does that mean and how can i fix it?