mholgatem/GPIOnext #65: Not working with an SPI connected joystick

KingWille, 2022-02-22:
I'm trying to use an analog joystick for my Retropie system, and I'm using an MCP3008 ADC via SPI to do it. The joystick reads just fine using my test codes, but when i have the Joystick connected to the GPIO pins, GPIOnext doesn't work. I get the following error:

RuntimeError: Failed to add edge detection

mholgatem, 2022-02-22:
@KingWille - Sorry, but GPIOnext doesn't handle SPI. It relies solely on the rpi.gpio library which only reads the states of the gpio pins, it doesn't handle any communication protocols like SPI or I2C. You could fork the code and add your own code using a library like spidev to get the values of your joystick. From there you could handle the joystick values in the [Axis Class](https://github.com/mholgatem/GPIOnext/blob/08379bedfd9fef7b9ba2e76ea30dccaf2a0c512c/config/device.py#L59). I know that this wasn't what you were hoping for, but unfortunately it's beyond the scope of what I designed gpionext for. 

Personally, I would look at using an Arduino Micro as a solution. It has analog pins and can emulate a gamepad via usb. You

closed 2022-02-22 by mholgatem

KingWille, 2022-02-23:
Ok, well thank you for your quick response and tips on how to solve it. Im a beginner working with a friend but I'll continue trying. We thought of the arduino solution and that's the backup plan, but ideally we want it to work with a ADC card.

joecoolish, 2024-02-06:
I just submitted a pull request that adds SPI support.  I have tested it with an MCP3008 and it works great! I would love additional testing.

mholgatem, 2024-02-06:
Awesome @joecoolish ! I'll check it out when I have a chance (it'll probably be a week or two). I'll merge it once I've had a chance to test it out. Thanks for the hard work!

KingWille, 2024-02-06:
Thanks mholgatem! Although I'm not working on this project anymore, thanks for coming with a solution. If you're interested, I have repo with spi functional code for gamepads in general. It's not my original code completely, just an altered version of another persons code. It's been awhile though, so I can't remember the creators name.