mholgatem/GPIOnext #67: GPIOnext doesn’t save configuration

Jackslayerr, 2022-03-30:
So I followed the steps to install GPIONEXT and after configuring the buttons when I go back to the Retropie screen it says it only sees 1 Gamepad which is the physical keyboard. Even when I use the test command it says it can’t find a configuration. The GPIO buttons work because it detects them when I map them but that’s it. I ran the command to start GPIOnext but no luck. There is no loose end in any GPIO configuration on my breadboard and they are all mapped to something. I’ve tried reinstalling the operating system but no luck. I’m using a Raspberry Pi 4 and I’ve unplugging my keyboard in times I could. This worked for me before without a problem and now it stopped working. What am I doing wrong?

Jackslayerr, 2022-03-31:
> Check your configuration file (SQLite database located in ../GPIOnext/config/config.db) upload it to an online SQLite database viewer like [this one](https://inloop.github.io/sqlite-viewer/) It should list all of your controllers + buttons + pins
> 
> If that looks good, then you'll need to make a video of the steps that you are going through to config your controller with gpionext. I just installed it on a fresh system today and didn't run into any problems.

I tried another fresh install and same issue. What do you mean by that directory. I only see the boot directory for my SD card and it doesn’t have any of those folders or files

mholgatem, 2022-04-01:
Where did you install GPIOnext to?

Jackslayerr, 2022-04-07:
I'm assuming the default install location. I just ran the steps from the Read me file. I had a formatted 32GB sd card that I flashed Retropie on. Then I installed GPIOnext

mholgatem, 2022-04-08:
ok, so it should be located in /home/pi/GPIOnext/config/config.db

Jackslayerr, 2022-04-08:
> ok, so it should be located in /home/pi/GPIOnext/config/config.db

Do I find this in the boot directory? It’s the partition that’s 2GB I can’t find any folder like that in there

mholgatem, 2022-04-08:
no, it's your root directory. if you use `ls -la` in your root directory, you will see several directories including both _boot_ and _home_

if you type `cd ~` that should set your current directory as _/home/pi_

Jackslayerr, 2022-04-09:
I cant get to that directory, is there anyway you can show me how to get to it? I typed the thing you said and nothing came up. I'm trying to view on Windows file explorer.

mholgatem, 2022-04-09:
Windows file explorer is probably only able to see very specific directories, unless you have configured it otherwise. You need to either do it on the pi itself or use something like putty or filezilla.

Jackslayerr, 2022-04-10:
So an update I’ve tried using different versions of Retropie but nothing helped. I tired to configure a virtual keyboard and this time when I went to configure input Retropie detected the gamepad in the configure input screen but the buttons don’t work. Now I really don’t know what’s going on

closed 2022-04-20 by mholgatem