mholgatem/GPIOnext #69: GPIOnext and sudo commands not working correctly

viniciusbrit, 2022-04-16:
I made a command to execute a python3 script with sudo. It locks up.
When I then type `sudo python3 script.py` it works flawlessly.

mholgatem, 2022-04-17:
I unfortunately have no solve for this. It has something to do with how systemctl services are run (for example, they don't see PATH variables). You will find that if you include the path to python in your command (`/usr/bin/python3 script.py`)and then just run gpionext as such from the command line
`sudo python3 ~/GPIOnext/gpionext.py`

your script likely will work (you'll notice that you don't need the 'sudo' in your command since the calling application already has sudo privileges. Unfortunately, I have spent many hours trying to fix this and I can't find anything that seems to work.  If you want to probe this more on your own, [this is a good place to start (though it doesn't seem to work)](https://stackoverflow.com/questions/53842351/python-subprocess-fails-to-find-shell-commands-when-started-by-systemctl)

mholgatem, 2022-04-17:
actually, there may be a fix. it appears that actually something is going on with my raspberry pi. I'lll have to troubleshoot and get back to you. In the meantime, you can try digging into adding the path variables, that may work for you.

mholgatem, 2022-04-17:
So I literally spent hours yesterday troubleshooting this, thinking that I had the answer. Turns out, I did. But, in previous testing, I had limited my pins to just pin 5. So everything that I tried, yielded no results 🙄. So yeah, I'm an idiot. I figured out my mistake and here's what you need to do. 

your command should be like this:
`/usr/bin/python3 /path/to/script.py`

you can either do a full path to your script, or a relative path from the gpionext folder. For instance, if you store your scripts in a folder called myscripts, which is stored inside the GPIOnext folder

`/usr/bin/python3 ./myscripts/script.py`

As stated above, you don't need to use the sudo command, since it already has elevated privileges.

closed 2022-04-17 by mholgatem

viniciusbrit, 2022-04-20:
Hi, thanks for clearing it up. 
I ended up just moving the GPIOnext functions to my original python program and using bash for other parts of it.