mholgatem/GPIOnext #70: Button combos not working

j4ym2, 2022-05-12:
Hi!
Thanks so much for your software its awesome.

I recently purchsed a gpio arcade stick and while your program works really well, i have noticed that I cannot do combos in games like final fight or street fighter (button 1+2+3  at the same time for special) 

I have played with the combo delay but i dont understand it all that well. I tried reading thropugh to see if there was any other issues related but i cant seem to find any. 

any help is appreciated.

mholgatem, 2022-05-12:
Sure, let's see what we can do. First off the combo delay works like this:

• You press a button -> a countdown timer starts
• The system logs all subsequent button pushes
• when the countdown finishes, if there are any combos present, it executes
those combos along with any remaining button presses

A few key things to note:
• Combo_delay is measured in milliseconds
• You have to set up your combo in gpionext config (ex. 'A' -> button1, 'B'
-> button2, 'C' -> buttons + button2)
• combos can be set up using any valid gpio pins, so you can include your
joystick as well (ex. up + button1)
• watch what pins gpionext config registers when setting up combos, you may
have an unexpected button accidentally being pressed, maybe even on player 2

If you still can't figure out what's going wrong, I'll be happy to help you
more, but it will have to wait till Monday. School finals and all 😅

j4ym2, 2022-05-13:
Thanks heaps - it can wait its not urgent :) 

I mean when playing the games - not a a pre set combo EG in street fighter - super move is qurter circle forward and 3 buttons, it never works. 

Final fight - special move is two buttons at the same time - some times works but not often. 

When i was using a usb encoder it seemed to work everytime. Any help like i said is appreciated :) good luck with finals!

mholgatem, 2022-05-13:
Ah, gotcha. You may not be able to do it with gpionext. Your best
chance would be to set combo_delay to 0 and make sure debounce is set to
something low like 1.

    Unfortunately, the python library rpi.gpio (and most of python itself)
is not suited for real time applications, there can be delays due to a
number of things. Add to that the threaded nature of gpionext, it's hard to
tell exactly when each key event will fire.

Hopefully that works for you

j4ym2, 2022-05-13:
Thanks so much for your input and feedback I'll give that a go.

closed 2022-06-02 by mholgatem