mholgatem/GPIOnext #71: Buttons not detected as gamepad

bjornthor21, 2022-05-17:
Hello I am trying to setup a little controller with 8 buttons, 4 for the dpad and the other 4 are (Start, Select, A, B), keep in mind this is a homebrew controller, i managed to go through the config and every button was detected and saved but when i boot up and i am asked to setup a gamepad it says "no gamepads detected" and it does not respond to anything from the buttons. 

thanks for your software it is amazing

###my setup:
> Raspberry pi model 3B+
> Retropie 4.8
> running on sd card
> pins: 29, 31, 33, 35, 37, 40, 38, 36 (every pin from 1 to 28 is occupied by agn lcd screen)

Thanks again for your software it is open source i really respect every single on that releases software open source! :),
let me know if you have any questions

mholgatem, 2022-05-18:
ok, couple of things to check first

1) make sure you start gpionext after config by running `gpionext start`
2) you can use jstest to test the controller by typing `gpionext test 1` (if you have multiple joysticks you can use 1,2,3,4 to test each)

If both of those are working, then its most likely a problem with a retropie config. In which case I would suggest trying to set up your controller as a gpionext keyboard and see if that works.

bjornthor21, 2022-05-18:
hello and thanks for the quick reply i did what you suggested but no luck. i reinstalled gpionext and went through the config again with with no success. i dont know if it is relevant but i run the config program with this command which you supplied sometime ago: sudo python3 /home/pi/GPIOnext/config_manager.py --pins 29,31,33,35,37,40,38,36  

i tried the "gpionext test 1" but i get this message: 
> cat: '/sys/devices/virtual/input/input*/name': No such file or directory
Could not find GPIOnext Joypad 1 in virtual devices

i then tried to setup the controller as a keyboard which did not seem to work

if it makes a diffrence i am connected to the pi using ssh

mholgatem, 2022-05-18:
I just realized, did you set the pins flag?
'gpionext set pins 29,31,33,35,36,37,38,40'

If not, it will error out when it gets to a pin that is already being used
by something else.

If it's not working after setting the pins, then it's probably hitting an
error somewhere. Stop gpionext by running:

'gpionext stop'

Then run it manually

'sudo python3 /home/pi/GPIOnext/gpionext.py --pins 29,31,33,35,37,40,38,36'

That should let you easily see what error you are getting

mholgatem, 2022-05-18:
I just realized, did you set the pins flag?
`gpionext set pins 29,31,33,35,36,37,38,40`

If not, it will error out when it gets to a pin that is already being used
by something else.

If it's not working after setting the pins, then it's probably hitting an
error somewhere. Stop gpionext by running:

`gpionext stop`

Then run it manually

`sudo python3 /home/pi/GPIOnext/gpionext.py --pins 29,31,33,35,37,40,38,36`

That should let you easily see what error you are getting

bjornthor21, 2022-05-18:
YESSSS! It worked and it is absolutely amazing SUPER COOL!. Thank you for your help it really helped me.

is there any way i can donate to you or this project? i would love. to it is fantastic to see how involved you are responding to issues and not stopping til it is solved keep up the good work.

mholgatem, 2022-05-19:
Hey that's great! I'm glad it's working for you! I really appreciate the enthusiasm for something I've made.

I don't currently take donations, but given what's going on in the world I think it would be great if you made a donation to [Refugees International](https://refugeesinternational.harnessapp.com/wv2/donate) or YOUR favorite charity!

closed 2022-05-19 by mholgatem