mholgatem/GPIOnext #73: Support for GPIO expanders?

yoboycal, 2022-06-10:
Hi, I'm using the GPIO expander found here [https://www.adafruit.com/product/4132] to add extra controls through I2C, but I can't get GPIOnext to recognize the pins. I know the pi is reading them after enabling I2C and has the 13 extra ones I'm using numbered from 41 to 53. I keep getting the "invalid channel" and "failed to add edge detection" errors. I tried to edit available pins in constants.py and also tried setting them using gpio set pins but neither of those worked. Is there a way to allow gpionext to read pins numbered after 40? Or is this a tall order? Thanks in advance

mholgatem, 2022-07-25:
sorry for the late reply, I thought that I responded to this already. Unfortunately, the library that GPIOnext uses (rpi.gpio) does not support I2C. So it's not something that will be implemented.

closed 2022-07-25 by mholgatem

mholgatem, 2022-10-11:
Hey there! I'm sorry to say that gpionext can't handle extra pins through
I2C. It uses the rpi.gpio python library which can't read I2C.

If you want to use gpionext, the best I can recommend is using a couple of
pins as shift registers along with a really short combo delay.