mholgatem/GPIOnext #75: 'dtoverlay=act-led,gpio=4' on makes "gpionext config" to fail

fedekrum, 2022-12-04:
Adding line `dtoverlay=act-led,gpio=4` to the /boot/config.txt makes `gpio config ` to fail. I use this to make gpio 4 an external led to know SD activity.

```
pi@retropie:~ $ gpionext set pins 12,16,18,22,32,36,37,38,40
setting pins 12,16,18,22,32,36,37,38,40
Setting Flag and Restarting GPIOnext...
pi@retropie:~ $ gpionext reload
pi@retropie:~ $ gpionext config
Traceback (most recent call last):
  File "/home/pi/GPIOnext/config_manager.py", line 315, in <module>
    ConfigurationManager(args)
  File "/home/pi/GPIOnext/config_manager.py", line 85, in __init__
    gpio.setupGPIO( self.args )
  File "/home/pi/GPIOnext/config/gpio.py", line 99, in setupGPIO
    pins.append( pin(pinNumber, pull, args) )
  File "/home/pi/GPIOnext/config/gpio.py", line 59, in __init__
    bouncetime = args.debounce)
RuntimeError: Failed to add edge detection
```
Any idea why ?

mholgatem, 2022-12-04:
yes, I know why, it's because I'm an idiot. I've been sitting on a fix for months that I thought I already uploaded. I'm not home right now, but I'll get it uploaded within a few hours of posting this comment. 
In the meantime, you can use the '--pins'  flag in conjunction with the gpionext config command to get what you are after
`gpionext config --pins 12,16,18,22,32,36,37,38,40`

fedekrum, 2022-12-04:
`yes, I know why, it's because I'm an idiot.` .... haaa haaaa !! You made me laugh !!!
I think that gpionex should not have "ALL" pins as a default and that it should force the user to define the pins first and should not run before it.
It will be your first correction in 4 years.... that is STABILITY !!!!

Thanks, I will wait for your update to do some tests

mholgatem, 2022-12-05:
Normally I would agree with you, but early on I tried something like that and it caused a lot more questions than it solved. 

Anyway, I just pushed the update, which I had to rewrite because I don't know what happened to it on my tester system, but it seems to have vanished. I probably deleted something thinking that it was already merged to github. lol. Anyway, I think that it was just one little part that I had changed. So let me know if you run into problems and I'll try to deal with it. I'll leave this issue open for a while and if you feel that it is adequate, you can close it.

fedekrum, 2022-12-05:
I will test tomorrow for sure...Thanks !!

fedekrum, 2022-12-05:
Ok ...we have a problem Houston !!

It gives me no error, but as sun as I do a gpionext config the led STOPS WORKING.
You are probably calling some other routine that does not check the pins set.

https://user-images.githubusercontent.com/4470683/205544765-2f922902-4bd3-403c-9acd-76ef2342569d.mp4

fedekrum, 2022-12-05:
> Normally I would agree with you, but early on I tried something like that and it caused a lot more questions than it solved.

Well... this is kind of a philosophical issue here.
For me there are 2 retro game users.

**a) The plug and play user.** 
This guys will buy a raspberry pi, copy the image, buy some plug and play controls (usb or BT) and go for it.

**b) The builder user.** 
This guy is the one implementing gpionext, so this guy has to get enough knowledge to understand that he MUST define the pins first and what a GPIO pin is.

So for me defining the pins that he is going to use will not only make it clearer to the builder, but you will also know where your responsibility for the GPIO pins end.

The config interface can show a first screen saying "Input the pins you are going to assign to GPIOnext delimited by commas". You can even validate if the pins are not valid (like pins 1 and 2 that are voltage, or ground pins) or the ones in use and alert him, and then proceed with the following step.

Just my thoughts !!

fedekrum, 2022-12-06:
Works!!
Thanks!!

fedekrum, 2022-12-06:
I am investigating a strange behaviour with some pins, I will report later

fedekrum, 2022-12-06:
One of my buttons (pin32-gpio12) is not detected by the config. As you see, is included in my 
` --pins 12,16,18,22,32,36,37,38,40`

```
raspi-gpio get 12
GPIO 12: level=0 fsel=4 alt=0 func=PWM0
```

if I install the old version I can be detected.

mholgatem, 2022-12-06:
GPIO 12: level=0 func=INPUT

fedekrum, 2022-12-06:
Thanks !!
I will investigate my particular case.

mholgatem, 2022-12-06:
No problem, let me know if you find the culprit!

fedekrum, 2022-12-07:
The culprit is me !! a typo at /boot/config.txt on the pwm audio. !!

mholgatem, 2022-12-07:
oops! Glad you figured it out!