mholgatem/GPIOnext #77: Install fails as per Jaycar instructions. Help. :)

glenhendry, 2023-01-01:
I can't install it using the instructions as it asks for username and password credentials and then fails because password authentication has been removed on aug last year?

mholgatem, 2023-01-01:
3 things (also, screenshots usually help):

What is asking for username and password?

What Raspbian release are you on?
Type `lsb_release -c` to find out.

Are you running as a user that has sudo privileges?
Type `getent group sudo` and check if your current username is listed in the output.

glenhendry, 2023-01-02:
[image]

[image]

Thanks for the clarifying questions. Sorry for the lack of detail.

Attached is the credentials request from the git clone and the answers to your other questions.

Thanks so much for your help.

Glen

mholgatem, 2023-01-02:
ah, I see the problem. You have a misspelling in the git address.
you added an extra 'e' in mholgatem
it should be `git clone https://github.com/mholgatem/GPIOnext.git`

closed 2023-01-02 by mholgatem

glenhendry, 2023-01-04:
Thanks! This fixed it. You're sw and help us appreciated.

mholgatem, 2023-01-04:
No problem!