mholgatem/GPIOnext #81: ModuleNotFoundError: No module named evdev

SerchBoogie, 2023-04-07:
Hi. 

I'm completely new to this. This device was already working with the arcade sticks and buttons soldered to the GPIO thingy, pins or something. I'm trying to fix this arcade stick for a family member who got it from a guy who is no longer available to repair it. 

My guess is that the memory got corrupted because it would no longer give video signal. only a black screen. So we bought a new SD card and installed RetrOrangePi v4.3, with Armbian 5.97. Everything works fine, but I still need to enable the arcade sticks and buttons soldered to the device. 

I Followed the instructions and it installs correctly I guess, but when I say yes to configure or type the command "gpionext config" I get the next error message: ModuleNotFoundError: No module named evdev. See screenshots. 

https://imgur.com/UnOVS0E
https://imgur.com/48HIDHL

I tried to type the command "gpionext start" but that does nothing and I cannot configure my controls using the config menu on the main UI. So basically the sticks and buttons won't respond. I can only use Keyboard or an Xbox Controller to navigate the system and play games. 

Help would be greatly appreciated.

merwok, 2023-04-07:
It seems that you haven’t run the install script that installs the evdev module: https://github.com/mholgatem/GPIOnext#how-to-install

naenaejeff, 2023-12-10:
Same problem, and reinstalling didnt fix? what am i doing wrong

mholgatem, 2023-12-10:
make sure you run the install script for GPIOnext
`bash ~/GPIOnext/install.sh`

If you have run the install script and it's still not working, you will want to run
`sudo pip3 install evdev`
and pay attention to the output to see why it hasn't installed correctly. I've never messed with orangepi, so I don't know how well evdev is supported.

naenaejeff, 2023-12-10:
sudo rm /usr/lib/python3.11/EXTERNALLY-MANAGED 
this fixed it, 
then 
apt pip3 install evdev

merwok, 2023-12-10:
That’s the kind of «fix» that will break your system later!

`pip install --user evdev` is the recommended way

closed 2024-03-08 by mholgatem