mholgatem/GPIOnext #82: gpionext config doesn't detect gpio inputs

xScotty, 2023-04-29:
Hello, I have a setup with 4 push buttons connected to the gpio of my rpi zero W.
I am able to retrieve the inputs with the RPi.GPIO python library and everything seems to work fine.

I installed `gpionext` by cloning the repo and running `bash GPIOnext/install.sh`.
When I run `gpionext config` and try to configure a controller, I'm stuck at the `Hold button` step and `gpionext` seems to not detect the inputs from my push buttons even after restarting `gpionext` or rebooting my pi.

ThatOneGuy140, 2024-01-19:
Same thing is happening to me I dont know if its something to do with the pins or not... But ill look into it further

DirectorOMR, 2024-02-29:
I'm also having this issue.  Raspberry Pi 5 Bookworm.

mholgatem, 2024-03-08:
@DirectorOMR, as you noted in issue #88 , The Raspberry pi 5 is not supported by the RPi.GPIO library. Whenever I can get my hands on one, I'll look into updating to a new library.  

@ThatOneGuy140 & @xScotty - have you found a solution to your problem? I have run it on both a raspberry pi zero W & zero 2 with no problems. The only thing that I can think of is if perhaps you tested the RPi.GPIO library using PULLDOWN (and are wired thusly). If that's the case, you can tell gpionext to use pulldown resistors with the command `gpionext set pulldown true`

closed 2024-03-08 by mholgatem