mholgatem/GPIOnext #83: Gpio next syntax error

Devenhulk123, 2023-07-20:
I wanted to use my raspberry gpio pins and I installed it using these scripts but I can't get pass this I dont know what to do or what I put to set can a input really need help

Thanks 
Deven

merwok, 2023-07-20:
Hello!  To get help with a problem on the internet, you need to say
- what command you run
- what error message you see (preferably as text not screenshots)

Devenhulk123, 2023-07-20:
> Hello! To get help with a problem on the internet, you need to say
> 
> * what command you run
> * what error message you see (preferably as text not screenshots)

Sorry Its my first time posting so I didn't really know what to put

mholgatem, 2023-07-21:
The reason that you are getting that error is that you are most likely running an old version of python3. Use this command to find your version:

`python3 --version`

Devenhulk123, 2023-07-21:
> The reason that you are getting that error is that you are most likely running an old version of python3. Use this command to find your version:
> 
> `python3 --version`
> 
> If your version is <3.6, upgrade it with: `sudo apt install python3.9`
I just tried and the same error popped up

merwok, 2023-07-21:
That’s probably because the script uses `python3`, and on your system that corresponds to Python 3.6, even if 3.9 is installed.
Can you update your operating system to a version that uses a current Python version?

Devenhulk123, 2023-07-21:
> That’s probably because the script uses `python3`, and on your system that corresponds to Python 3.6, even if 3.9 is installed. Can you update your operating system to a version that uses a current Python version?

I used sudo apt-get update followed by sudo apt-get upgrade  and everything thing is updated already what do you mean by upgrade the operating system?

merwok, 2023-07-21:
Answering that would take some time, and be out of scope for this place!

But can you show the output of these commands:
```
cat /etc/os-release
apt-cache show python3 | head -n3
```

Devenhulk123, 2023-07-24:
> Answering that would take some time, and be out of scope for this place!
> 
> But can you show the output of these commands:
> 
> ```
> cat /etc/os-release
> apt-cache show python3 | head -n3
> ```

Sorry for the late reply I went on a trip with my parents and we didn't have any signal if your still willing to help these are the stuff that popped up

merwok, 2023-07-24:
Your link goes to this same page.
Can you copy-paste the output from the two commands?

Devenhulk123, 2023-07-24:
> Your link goes to this same page. Can you copy-paste the output from the two commands?

pi@retropie:~$ cat /etc/os-release

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)'

NAME="Raspbian GNU/Linux"

VERSION ID="9"

VERSION="9 (stretch)"

ID=raspbian

ID LIKE-debian

HOME URL="http://www.raspbian.org/"

SUPPORT URL="http://www.raspbian.org/RaspbianForums" BUG REPORT URL="http://www.raspbian.org/RaspbianBugs"

pi@retropie:~$ apt-cache show python3 | head -n3

Package: python3

Source: python3-defaults

Version: 3.5.3-1

pi@retropie:~ $

Devenhulk123, 2023-07-24:
> Your link goes to this same page. Can you copy-paste the output from the two commands?

Sorry that was my bad

merwok, 2023-07-24:
We learn two things:
- you are running raspberry pi OS 9, but the current version is 10
- the python version used by the `python3` command is 3.5, which is too old to support this script

Raspberry Pi foundation does not recommend updating the system in place, but making a fresh install on a new SD card.

Devenhulk123, 2023-07-24:
> We learn two things:
>
> * you are running raspberry pi OS 9, but the current version is 10
> * the python version used by the `python3` command is 3.5, which is too old to support this script
>
> Raspberry [...]

Is there any way I could try updating it without starting new SD card before I wipe it clean

mholgatem, 2023-07-24:
Yes, you can. But in my experience, you tend to get better performance from a fresh install. If you are determined to just upgrade in place, [here's a guide](https://www.makeuseof.com/tag/raspberry-pi-update-raspbian-os/)

Devenhulk123, 2023-07-24:
> Yes, you can. But in my experience, you tend to get better performance from a fresh install. If you are determined to just upgrade in place, [here's a guide](https://www.makeuseof.com/tag/raspberry-pi-update-raspbian-os/)

Okay thank you I'll try it out

closed 2024-03-08 by mholgatem