mholgatem/GPIOnext #87: inputs not working p0W

2kohm, 2024-02-27:
Thank you for providing this excellent tool! However, it's also been a bit of a challenge, but in a humorous way :D Here's the breakdown of my current test setup:

Raspberry Pi Zero W with the vanilla RetroPie image, no modifications.
Integration of keys using gpioNEXT with a tactile PCB containing 12 buttons.
I successfully installed and configured everything, and initially, the buttons were working flawlessly (confirmed with gpionext test 1). Then, I decided to enhance my setup by enabling my ILI9341 320x240 TFT display via SPI1 or 0. Following a reboot, the screen was running smoothly at 60 FPS, but unfortunately, the buttons ceased to function within EmulationStation.

Moreover, I encountered issues with PWM audio on GPIO 12, 13, 18, which not only caused the buttons to exhibit erratic behavior but also resulted in significant FPS drops after the fourth attempt to reinstall everything

mholgatem, 2024-03-08:
Did you use the ` set pins ` command? 
example: `gpionext set pins 3,5,38,40`

Otherwise gpionext grabs all of the pins, which can have unpredictable consequences.

mholgatem, 2024-03-08:
I should elaborate. Use the `set pins` command to tell gpionext to only look at the specific pins that your controller uses.

closed 2024-09-26 by mholgatem