mholgatem/GPIOnext #91: Exception in thread

amak79, 2024-06-03:
GPIOnext crashes with the following error:
```
Jun 01 22:23:13 retropie python3[1153]: Exception in thread Thread-2138:
Jun 01 22:23:13 retropie python3[1153]: Traceback (most recent call last):
Jun 01 22:23:13 retropie python3[1153]:   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
Jun 01 22:23:13 retropie python3[1153]:     self.run()
Jun 01 22:23:13 retropie python3[1153]:   File "/usr/lib/python3.7/threading.py", line 1166, in run
Jun 01 22:23:13 retropie python3[1153]:     self.function(*self.args, **self.kwargs)
Jun 01 22:23:13 retropie python3[1153]:   File "/home/pi/GPIOnext/config/device.py", line 281, in processQueue
Jun 01 22:23:13 retropie python3[1153]:     threading.Thread( target=currentEvent.press ).start()
Jun 01 22:23:13 retropie python3[1153]:   File "/usr/lib/python3.7/threading.py", line 847, in start
Jun 01 22:23:13 retropie python3[1153]:     _start_new_thread(self._bootstrap, ())
Jun 01 22:23:13 retropie python3[1153]: RuntimeError: can't start new thread
```
Full log: [gpionext.log](https://github.com/user-attachments/files/15529572/gpionext.log)

I first experienced this after about 5 mins of playing the shoot 'em up M.U.S.H.A. for Genesis, and it keeps happening every time I play it. This game has auto fire so you only need to keep a button pressed to fire. I can reproduce this even when not in the game by keeping a button pressed and randomly moving the joystick.

I might be related to https://github.com/mholgatem/GPIOnext/issues/43, but I don’t seem to hit the thread cap. The error usually occurs somewhere between 2000 and 5000 threads.

**Hardware:** Raspberry Pi 4B 1 GB
**OS:** RetroPie 4.8 (fully updated)
**GPIOnext:** Latest git with default options

mholgatem, 2024-06-03:
Did you try setting combo_delay to 0?

amak79, 2024-06-04:
I did set `combo_delay` to `0` but the same error occurs.

mholgatem, 2024-06-05:
How long does it usually take for you to get this error?
While you are playing, if you release the button for _half a second_ every once in a while, does it still have the problem?

amak79, 2024-06-06:
The error usually occurs after 5 mins.

I did a test where I would release the button every 30 seconds for half a second and the error didn't occur. I ran the test for 6 mins. The `combo_delay` was set to the default of `50`.

Is there any way to list the current config values?

mholgatem, 2024-06-13:
sorry, I've been away for a bit. The config variables are set as flags in the service file. You can open `/lib/systemd/system/gpionext.service` to see all of the currently set flags