mholgatem/GPIOnext #92: Adding A Button Which You Cannot Press

KieranReck, 2024-06-13:
Hi all, apologies if this is the wrong way to ask this question, i'm new here.

How would I go about adding an additional button to my GPIOnext configuration without being able to press it? The 'button' is actually a circuit which gets pulled low when the device's battery is discharged, but this makes it very difficult to configure using the interactive configuration because I cant easily trigger it without the battery actually going low.

I had a look in the GPIOnext folder to see if I could find where my existing gpio 'device' config is stored but I couldn't find it.

Any tips would be appreciated!

mholgatem, 2024-06-13:
There are two ways that you could go about doing it. 

1.  Wire a temporary button (or just short the pin temporarily) so that you can configure it
2.  Manually edit the sqlite database /home/pi/gpionext/config/config.db

for # 2 you would use a command similar to this (add your own values):
```
cd ~/gpionext/config
sqlite3 config.db
insert into GPIOnext values('Commands', 'Low Battery', 'COMMAND', 'shutdown now;', '7');
```
the values are 'device type' (Leave as Commands), 
'name of command' (Low Battery), 
'COMMAND' (Leave this), 
'semi-colon separated list of commands to run' (shutdown now), 
pin (comma separated if using multiple pins)

commands are run with this python code within gpionext:
`subprocess.call( [cmd], executable='/bin/bash', shell=True`

closed 2024-06-13 by mholgatem

KieranReck, 2024-06-30:
Hi there, thanks for your response, I finally found time to try it out. I am getting a response from sqlite stating that the table has 6 columns but only 5 values were received. Is there supposed to be an additional value in the string in your answer?

mholgatem, 2024-06-30:
Forgive me, The sixth value (which is actually column 1) would probably be 'unique ID', which should be auto generated, but maybe the command line handles it differently? Try something like this:
`insert into GPIOnext values(999, 'Commands', 'Low Battery', 'COMMAND', 'shutdown now;', '7')`

KieranReck, 2024-07-09:
Perfect, that works! Command behaves as expected, although unfortunately I have now found out that my circuit gives a fluctuating signal due to battery voltage sag under use. Easiest way to solve this will probably be to add a long (multiple second) debounce to that input specifically. Not sure whether that is in the scope of gpionext, or if I will need to write a separate watch for that pin.

Either way, thanks for your help, much appreciated!

----

For anyone attempting this in the future, here's the overall process:

```
cd ~/gpionext/config
sqlite3 config.db
insert into GPIOnext values(98, 'Commands', 'Low Battery', 'COMMAND', 'sleep3 && shutdown now;', '29');
```
note: if the unique ID you chose is already occupied, you will receive an error, just try a different number.

exit sqlite3. If all went well, after a reboot you will now be able to see your commandv listed in the gpionext configurator under the commands section :)

----
Additional info from earlier in the chain still applies:
The values in the sqlite entry are:
- Unique ID (just pick a number that isnt used, if it is used, it will tell you in an error)
- 'device type' (Leave as Commands),
- 'name of command' (Low Battery),
- 'COMMAND' (Leave this),
- 'semi-colon separated list of commands to run' (sleep3 && shutdown now),
- pin (comma separated if using multiple pins)

commands are run with this python code within gpionext:
subprocess.call( [cmd], executable='/bin/bash', shell=True