mholgatem/GPIOnext #93: "Can't add edge detection for pin" "(pin is already in use)" errors until python3-rpi-lgpio installed

icculus, 2024-06-16:
Hi, this is _not_ a bug in GPIOnext, so this can be closed immediately, but I want to save the next person that searches the issue tracker, like I did, some time.

The latest Raspberry PI OS (based on Debian "Bookworm") will absolutely _not work_ with GPIOnext if you have the usual Python3 RPi.GPIO package installed. The daemon and config program will both run, but GPIOnext's logs will write things like "Can't add edge detection for pin 5(pin is already in use). Skipping." and they just won't ever see any activity on any GPIO pin (so when you get to the "hold button" step of the config program, it just won't ever see the button press).

This happens because the usual RPi.GPIO package doesn't work with the new kernel interfaces in Bookworm. You need to uninstall that package and install the "lgpio" version, which provides the same Python interface as RPi.GPIO but talks to the new kernel interfaces behind the scenes.

So do this:

```bash
sudo apt remove python3-rpi.gpio
sudo apt update
sudo apt install python3-rpi-lgpio
```

And maybe reboot, and then GPIOnext (and everything else that's using this Python package) will magically stop giving weird errors and start working.

This was a full day of me pulling my hair out in frustration, so hopefully someone stumbles upon this and it helps them.  :)

icculus, 2024-06-16:
(if there's an action to take before closing this, it's maybe add a note to the README about this, but otherwise I just wanted to make something people will find in searching the issue tracker. Thanks for this awesome project! Once I figured out what broke in the latest Raspberry Pi OS, it works like a dream!)

mholgatem, 2024-06-20:
I updated the installer to check for bookworm or later and install accordingly. I don't have time to set up a system and test the installer, so if you'd be willing to test it out, that would be great. The new installer should tell you the debian version # and name during installation as well as installing rpi.gpio or rpi-lgpio as necessary. 

Thanks for your input. Please let me know if you have a chance to test the new installer. I'll leave this issue open for the time being, if anyone tests it, please leave a comment.

icculus, 2024-06-20:
I had already sealed up the previous Pi in a game console thing, so I put together a Pi Zero with a fresh Bookworm install of Raspberry Pi OS, and present to you the world's fastest-made--and worst--GPIO joystick:

[image]

install.sh definitely uninstalled the usual RPi.GPIO module and installed the new (correct) one. 

I'm honestly surprised the OS ships with the old package installed by default, as it shouldn't work on this OS at all now. Presumably they'll fix that at some point...?

It also needs `python3-evdev` installed through apt, as that's not installed by default and the GPIOnext config program won't start without it...in case you want to throw that in, too.

With that out of the way, the config program saw all four buttons on this breadboard without any issue, so I think we're good to go here!

icculus, 2024-06-20:
Also tried on the latest RetroPie (based on Buster), and it worked correctly there too, so no regression.

Notably, I saw install.sh install the evdev module here. I think there was a warning about not installing something globally/outside a venv on Bookworm, which was probably that, but I wasn't paying attention. Since you're installing from apt anyhow, you might as well just use the apt package for it.

mholgatem, 2024-06-20:
Good catch, thanks for testing that. I updated to apt. I should be able to remove the python3-pip dependency, but I'm doing everything on my phone right now and I want to double check when I get back to my desktop. 

Thanks again for testing!

closed 2024-06-20 by mholgatem