mholgatem/GPIOnext #95: Joypad 1 not recognized after launching game, works in retropie menu

willelind, 2024-08-23:
Hi!

Hardware: Pi Zero 2
OS: RetroPie 4.8

I did a config via instructions from this GitHub, then I used that config to create a gamepad via emulationstation config. This works as intended and the pins from joypad1 are recognized as a game pad. I can use all the keys as intended in the retropie menu.

However, when I boot up any game, the keys stop being responsive and I get the following error on the screen:

GPIOnext Joypad 1 (9999/8888) not configured

What can I do to solve this?

mholgatem, 2024-08-23:
Hmm. Unsure. This seems like it's a problem on the emulationstation side of
things if it's working in the retropie menu.

willelind, 2024-08-23:
Interesting, thanks for quick response.

The weird thing is, if I create a joypad 1 config in gpionext where I bind for example pin 40 to A, and then I bind that button in emulationstation to B, it will act as B in the retropie menu. Which seems to be indicating that the gamepad config from emulationstation works. But when launching the game it stop working.

And to add more problem, I can see that the device from:

/sys/devices//.../inputs/ is removed when launching a game. So it seems to me that it tries to use the input device from gpionext and deletes it because it can't.

I didn't get any extra info while running with debug flag.

mholgatem, 2024-08-23:
Correct. The bindings in gpionext are the device BINDINGS whereas in
retropie they are MAPPINGS. (I.E. - 'B' » 'A')

mholgatem, 2024-08-23:
Sorry I couldn't be more help

willelind, 2024-08-23:
Thank you all the same, for the quick response nonetheless. I also want to thank you for a great tool, it really helped me a lot during setup for my quite complex project (a keyboard that can switch state between controlling the computer or controlling the embedded emulator).

All the same, I did figure it out finally, so I wanted to post a response here in case anyone else gets this problem.

9999/8888 seems to be related to joystick/dpad. I did find an old thread https://retropie.org.uk/forum/topic/32997/controller-working-in-menu-but-not-in-game/2 about this, and it seems if you use a joystick/dpad you can add your 9999/8888 or whatever numbers you have into your config file.

The problem/solution for me is that I don't have a dpad, so if I don't register one and simply enter my pins as a keyboard in gpionext config instead, it works as intended.

mholgatem, 2024-08-23:
Great, I'm glad you were able to get it working!

closed 2024-09-26 by mholgatem