mholgatem/GPIOnext #97: Any way to add matrix buttons?

LaSaN-101, 2024-12-04:
Running out of gpio for a full controller, need to make matrix to reduce pins, didn't find anyway in readme, cant read the scripts very well, please help. Thank you.

mholgatem, 2024-12-05:
That is what the combos/combo delay are for. You can press two or more buttons at once ( or tie one button to two gpio pins) during config.

LaSaN-101, 2024-12-11:
wait am i understanding it wrong? i want to tie multiple buttons to one gpio (essentially) like how keyboards work, but combos are like macros, doing multiple keypresses with one key press.

mholgatem, 2024-12-11:
no, combos with GPIOnext are matrix buttons. 
You can map Button 1 to gpio 5, Button 2 to gpio 6, and then map Button 3 to Button 1 + Button 2 at the same time. 

You can also use this in reverse with different wiring schemes. For instance: 
if gpio 5 only -> fire Button 1
if gpio 6 only -> fire Button 2
if gpio 7 only -> fire Button 3
if gpio 5 + gpio 6 -> fire Button 4
if gpio 5 + gpio 7 -> fire Button 5
if gpio 6 + gpio 7 -> fire Button 6

this is just a basic example, you can create more complicated wiring schemes to get what you are looking for. 

Honestly though, if you need substantially more buttons that the 26 gpio, I would look into alternative solutions such as using an arduino pro micro as a usb hid device (you can get cheap knockoffs for ~$5).

LaSaN-101, 2024-12-11:
ohk i get it now, so i was understanding it wrong, thank you for the clarification. I might use this, but i have already made a hid controller using the rp2040, but too has limited analog pins. so might as well use only one board and use this library.